openfans-community-offical/Debian-Pi-Aarch64 #3: 请问armhf本地安装包如何安装呢？

Purek, 2019-02-06:
我尝试这样运行
sudo apt install ./armhf.deb:armhf
sudo dpkg -i armhf.deb
均提示依赖错误，但是依赖包我安装它，提示已经安装过了

pi@raspbian:~$ sudo dpkg -i armhf.deb
(正在读取数据库 ... 系统当前共安装有 125396 个文件和目录。)
正准备解包 armhf.deb  ...
正在将 teamviewer-host:armhf (12.1.83885) 解包到 (12.1.83885) 上 ...
dpkg: 依赖关系问题使得 teamviewer-host:armhf 的配置工作不能继续：
 teamviewer-host:armhf 依赖于 libc6 (>= 2.17)；然而：
  未安装软件包 libc6:armhf。
 teamviewer-host:armhf 依赖于 libdbus-1-3；然而：
 teamviewer-host:armhf 依赖于 libqt5gui5 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 libqt5widgets5 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 libqt5qml5 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 libqt5quick5 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 libqt5webkit5 (>= 5.1.1)；然而：
 teamviewer-host:armhf 依赖于 libqt5x11extras5 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 qtdeclarative5-qtquick2-plugin (>= 5.2.1) | qml-module-qtquick2 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 qtdeclarative5-controls-plugin (>= 5.2.1) | qml-module-qtquick-controls (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 qtdeclarative5-dialogs-plugin (>= 5.2.1) | qml-module-qtquick-dialogs (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 qtdeclarative5-window-plugin (>= 5.2.1) | qml-module-qtquick-window2 (>= 5.2.1)；然而：
 teamviewer-host:armhf 依赖于 qtdeclarative5-quicklayouts-plugin (>= 5.2.1) | qml-module-qtquick-layouts (>= 5.2.1)；然而：

dpkg: 处理软件包 teamviewer-host:armhf (--install)时出错：
 依赖关系问题 - 仍未被配置
正在处理用于 lastore-daemon (0.9.55-1) 的触发器 ...
正在处理用于 desktop-file-utils (0.23-2) 的触发器 ...
正在处理用于 bamfdaemon (0.5.3-2+b1) 的触发器 ...
Rebuilding /usr/share/applications/bamf-2.index...
正在处理用于 mime-support (3.60) 的触发器 ...
正在处理用于 hicolor-icon-theme (0.17-1) 的触发器 ...
在处理时有错误发生：
 teamviewer-host:armhf

sudo apt install libqt5quick5:armhf
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
您也许需要运行“apt --fix-broken install”来修正上面的错误。
下列软件包有未满足的依赖关系：
 libqt5quick5 : 破坏: libqt5quick5:armhf (!= 5.6.1-11) 但是 5.7.1-2+b2 正要被安装
 libqt5quick5:armhf : 依赖: libc6:armhf (>= 2.4) 但是它将不会被安装
                      依赖: libgcc1:armhf (>= 1:3.5) 但是它将不会被安装
                      依赖: libgles2-mesa:armhf (>= 7.8.1) 但是它将不会被安装 或
                              libgles2:armhf
                      依赖: libqt5core5a:armhf (>= 5.7.0) 但是它将不会被安装
                      依赖: libqt5gui5:armhf (>= 5.7.0) 但是它将不会被安装
                      依赖: libqt5network5:armhf (>= 5.0.2) 但是它将不会被安装
                      依赖: libqt5qml5:armhf (>= 5.7.1) 但是它将不会被安装
                      依赖: libstdc++6:armhf (>= 5) 但是它将不会被安装
                      依赖: qtbase-abi-5-7-1:armhf
                      依赖: qtdeclarative-abi-5-7-0:armhf
                      破坏: libqt5quick5 (!= 5.7.1-2+b2) 但是 5.6.1-11 正要被安装
 teamviewer-host:armhf : 依赖: libc6:armhf (>= 2.17) 但是它将不会被安装
                         依赖: libdbus-1-3:armhf 但是它将不会被安装
                         依赖: libqt5gui5:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: libqt5widgets5:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: libqt5qml5:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: libqt5webkit5:armhf (>= 5.1.1) 但是它将不会被安装
                         依赖: libqt5x11extras5:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: qtdeclarative5-qtquick2-plugin:armhf (>= 5.2.1) 但无法安装它 或
                                 qml-module-qtquick2:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: qtdeclarative5-controls-plugin:armhf (>= 5.2.1) 但无法安装它 或
                                 qml-module-qtquick-controls:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: qtdeclarative5-dialogs-plugin:armhf (>= 5.2.1) 但无法安装它 或
                                 qml-module-qtquick-dialogs:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: qtdeclarative5-window-plugin:armhf (>= 5.2.1) 但无法安装它 或
                                 qml-module-qtquick-window2:armhf (>= 5.2.1) 但是它将不会被安装
                         依赖: qtdeclarative5-quicklayouts-plugin:armhf (>= 5.2.1) 但无法安装它 或
                                 qml-module-qtquick-layouts:armhf (>= 5.2.1) 但是它将不会被安装
E: 有未能满足的依赖关系。请尝试不指明软件包的名字来运行“apt --fix-broken install”(也可以指定一个解决办法)。

aiminickwong, 2019-02-07:
U should use  "apt install" for all packages and add ":armhf" as a suffix, sometimes solved for lib class of armhf pkg,  if U wanna let userland program except lib files to 64 & 32 together, the better way is used chroot as an only 32bit env.

closed 2019-02-08 by aiminickwong

Jayjjjjj, 2020-12-06:
same problem... 
` sudo apt install ./synergy_1.11.0.stable_b105+b2173efb_raspbian_armhf.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'synergy:armhf' instead of './synergy_1.11.0.stable_b105+b2173efb_raspbian_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 synergy:armhf : Depends: libqt5gui5:armhf (>= 5.0.2) but it is not going to be installed
                 Depends: libqt5widgets5:armhf (>= 5.2.0~alpha1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`
So I used aptitude to fix dependency problems.
`The following packages have unmet dependencies:
 libegl-mesa0 : Breaks: libegl-mesa0:armhf (!= 19.1.6-1) but 18.3.6-2+deb10u1 is                                                  to be installed
 libegl-mesa0:armhf : Breaks: libegl-mesa0 (!= 18.3.6-2+deb10u1) but 19.1.6-1 is                                                  installed
 libglapi-mesa : Breaks: libglapi-mesa:armhf (!= 19.1.6-1) but 18.3.6-2+deb10u1                                                  is to be installed
 libglapi-mesa:armhf : Breaks: libglapi-mesa (!= 18.3.6-2+deb10u1) but 19.1.6-1                                                  is installed
 libgbm1 : Breaks: libgbm1:armhf (!= 19.1.6-1) but 18.3.6-2+deb10u1 is to be ins                                                 talled
 libgbm1:armhf : Breaks: libgbm1 (!= 18.3.6-2+deb10u1) but 19.1.6-1 is installed
open: 71; closed: 224; defer: 51; conflict: 75`
Really dont know what to do next..