openfans-community-offical/Debian-Pi-Aarch64 #8: 反馈个TF卡支持问题

Miinec, 2019-02-15:
我尝试安装了，除开xface的其他所有版本，均会出现UEFI install shell界面，忘了截图，如果需要我可以补图，在出现大的树莓派LOGO后，进度加载完成后出现的。
大概提示分区BLK1，BLK2，BLK3，提示starup.nsh相关信息，之后就是shell>的交互界面。

一张TOSHIBA 32G UHS-II U3 C10的卡具体型号是M501G0320C7，可以兼容UHS-II HD312和UHS-I SDR104模式，这些参数我不是很明白，作为信息一起提供。
这张卡我尝试装了官方的reaspbian镜像，没有问题，能够正常进入桌面。

我用了另一张金士顿的 16G SDHC-I C10 U1 装了无桌面版，可以正常使用。

aiminickwong, 2019-02-16:
@Miinec 

谢谢你的反馈，你可以尝试以下3种办法

- 这张卡我也有，兼容性不是很好，有时候需要多烧录几次才可以。

- 如果出现 **UEFI shell**，请点击**ESC**进入**BIOS设置**，更改启动选项，从SD开启动，记得选择F10保存并在重启后再次进入BIOS检查。

- 复制 **/boot/EFI/BOOT/bootaa64.efi** 到 /boot目录下，改名为 **kernel8.img** 修改 **config.txt** 配置，将 **armstub=RPI_EFI.fd** 改为 **armstub=kernel8.img** 或 **kernel=kernel8.img**

Miinec, 2019-02-16:
@aiminickwong 

十分感谢你的解答~

aiminickwong, 2019-02-17:
@Miinec 初步估计是卡的问题
我将保持这个问题开放，欢迎随时反馈。

aiminickwong, 2019-02-25:
@Miinec Please use the newest version: U8, and TF card to boot your system.

closed 2019-02-25 by aiminickwong