openfans-community-offical/Debian-Pi-Aarch64 #15: 树莓派3B，做好系统以后卡在starting Reboot

small32, 2019-07-22:
第一次重启，系统是V1.0稳定版

aiminickwong, 2019-07-22:
请提供具体的信息，1.0也有很多版本，以及具体的截图等信息

small32, 2019-07-22:
2019-07-12-OPENFANS-Debian-Stretch-Aarch64-ext4-v2019-1.0-U10.img.xz

这个版本，第一次重启正常，第二次重启卡Starting Reboot，用了三星的卡和金士顿的卡，都是16G U1的卡，都卡这里，等了半个多小时都是这样

small32, 2019-07-22:
我用的显示器是4K分辨率的，是这个问题嘛？不支持4K？

small32, 2019-07-22:
感觉应该和显示分辨率无关，官方系统是正常运行的

small32, 2019-07-22:
试了下国外大神的Pi64，一切正常

aiminickwong, 2019-07-22:
[image]

@small32 您好，非常感谢您的反馈，我们也在第一时间使用**不同的3块** 【树莓派**3B** 】来复现这个问题，但是很遗憾的是我们没有遇到您说描述的情况。

目前可以肯定这个问题与显示（分辨率）及系统都没有关系，**如上图所示，这是我们在3B上测试的结果，系统功能一切正常**。

建议首先核对一下所下载的文件**MD5**是否正确并**重新正确烧录**系统镜像再进行测试。

再次感谢您的反馈。

aiminickwong, 2019-07-22:
此问题与所发行的系统镜像无关，鉴于方便用户反馈，暂时不关闭该问题。

small32, 2019-07-23:
奇怪，我机器根本不显示你这个界面

这么说吧，你们的系统刷入后据说会重启三次，第一次正常可以重启，第二次检测设备，有一个fail，是蓝牙，然后到最后重启，卡死在starting reboot，一直在这里卡住不动，无法继续，如果拔掉电源，再插，启动后还是检测设备然后到starting reboot卡死，不知道为毛

我机器现在刷Pi64稳定运行，懒得折腾了，等我啥时候再弄个树莓派再试试你们的系统吧

aiminickwong, 2019-07-23:
您好，您提供的信息无法便于我们继续跟进此问题，我们再次确认3B上测试一切正常，鉴于目前情况，将该问题关闭。

closed 2019-07-23 by aiminickwong