openfans-community-offical/Debian-Pi-Aarch64 #18: mirror download link ?

Akiafroo, 2019-08-07:
its not possible to download from baidu ,- can you upload those files in a different server ? dropbox, mega, etc,etc, or more easy just upload to github as release.

aiminickwong, 2019-08-07:
@Akiafroo  It's in our plan, but very sorry not ye now.

hubba1987, 2019-08-25:
> its not possible to download from baidu ,- can you upload those files in a different server ? dropbox, mega, etc,etc, or more easy just upload to github as release.

I've spent half a day downloading from Baidu, but now I can share the images from RC2 branch:
https://1drv.ms/u/s!AnqNjFzt3UZx0Cetgpj705linXAF?e=BE8YHI

Owner, pls, tell me if I'm not allowed to share.

aiminickwong, 2019-09-26:
@hubba1987   you can share anywhere, But I suguest you Upload a readme file to tell people where from and who built it. :)

closed 2019-09-26 by aiminickwong