openfans-community-offical/Debian-Pi-Aarch64 #29: Monitorix异常

bmyjacks, 2019-08-30:
在安装了Monitorix之后，配置文件中添加对树莓派的支持，但是在监控界面中关于树莓派的显示信息为空白，希望能解决这个问题。

aiminickwong, 2019-08-30:
日志。

bmyjacks, 2019-08-30:
我能上传图片吗

bmyjacks, 2019-08-30:
然后再上传日志

bmyjacks, 2019-08-30:
[image]

bmyjacks, 2019-08-30:
Fri Aug 30 21:35:53 2019 - Starting Monitorix version 3.10.1 (pid 2645).
Fri Aug 30 21:35:53 2019 - Loaded main configuration file '/etc/monitorix/monitorix.conf'.
Fri Aug 30 21:35:53 2019 - Loading extra configuration file '/etc/monitorix/conf.d/00-debian.conf'.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
Fri Aug 30 21:35:54 2019 - Initializing graphs.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/system.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/kern.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/proc.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/fs.rrd' file.
Fri Aug 30 21:35:54 2019 - fs::fs_init: Unable to detect the device name of '/', I/O stats won't be shown in graph. If this is really a mount point then consider using <devmap> to map it manually to a device name.
Fri Aug 30 21:35:54 2019 - fs::fs_init: Unable to detect the device name of 'swap', I/O stats won't be shown in graph. If this is really a mount point then consider using <devmap> to map it manually to a device name.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/net.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/netstat.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/serv.rrd' file.
Fri Aug 30 21:35:54 2019 - Creating '/var/lib/monitorix/port.rrd' file.
Fri Aug 30 21:35:55 2019 - Creating '/var/lib/monitorix/user.rrd' file.
Fri Aug 30 21:35:55 2019 - Creating '/var/lib/monitorix/int.rrd' file.
Fri Aug 30 21:35:55 2019 - Generating the 'index.html' file.
Fri Aug 30 21:35:55 2019 - Started built-in HTTP server (pid 2797).
Fri Aug 30 21:35:55 2019 - Ok, ready.
Fri Aug 30 21:35:55 2019 - WARNING: the built-in HTTP server has authentication disabled.
HTTPServer: You can connect to your server at http://localhost:8080/
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = "en_US:en",
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = ""
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Argument "IPI0" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 15.
Argument "IPI1" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 16.
Argument "IPI2" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 17.
Argument "IPI3" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 18.
Argument "IPI4" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 19.
Argument "IPI5" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 20.
Argument "IPI6" isn't numeric in numeric gt (>) at /usr/lib/monitorix/int.pm line 876, <IN> line 21.
Argument "IPI0" isn't numeric in numeric lt (<) at /usr/lib/monitorix/int.pm line 1018.
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fontconfig warning: ignoring UTF-8: not a valid region tag
Fri Aug 30 22:05:00 2019 - ERROR: while updating /var/lib/monitorix/proc.rrd: could not lock RRD
Fri Aug 30 22:06:57 2019 - SIGTERM caught.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
Fri Aug 30 22:06:58 2019 - Exiting.
Fri Aug 30 22:06:58 2019 - Starting Monitorix version 3.10.1 (pid 14277).
Fri Aug 30 22:06:58 2019 - Loaded main configuration file '/etc/monitorix/monitorix.conf'.
Fri Aug 30 22:06:58 2019 - Loading extra configuration file '/etc/monitorix/conf.d/00-debian.conf'.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
Use of uninitialized value in lc at /usr/lib/monitorix/disk.pm line 60.
Fri Aug 30 22:06:58 2019 - Initializing graphs.
Fri Aug 30 22:06:59 2019 - disk::disk_init: ERROR: invalid or inexistent device name '/dev/sda'.
Fri Aug 30 22:06:59 2019 - disk::disk_init: 'accept_invalid_disk' option is not set.
Fri Aug 30 22:06:59 2019 - disk::disk_init: WARNING: initialization aborted.
Fri Aug 30 22:06:59 2019 - fs::fs_init: Unable to detect the device name of '/', I/O stats won't be shown in graph. If this is really a mount point then consider using <devmap> to map it manually to a device name.
Fri Aug 30 22:06:59 2019 - fs::fs_init: Unable to detect the device name of 'swap', I/O stats won't be shown in graph. If this is really a mount point then consider using <devmap> to map it manually to a device name.
Fri Aug 30 22:06:59 2019 - Creating '/var/lib/monitorix/lighttpd.rrd' file.
[... 214 lines cut ...]
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
[... cut ...]
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory

aiminickwong, 2019-08-31:
看到错误：/opt/vc/bin/vcgencmd: error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory

应该使用/opt/vc64/bin/vcgencmd 
libvchiq_arm.so在/opt/vc64/lib/目录下

这个不是问题，需要你执行设置环境变量
@bmyjacks

closed 2019-08-31 by aiminickwong