openfans-community-offical/Debian-Pi-Aarch64 #51: VNC can not be installed

Jyurineko, 2019-10-17:
i install the macOS Mojave theme custom desktop (Desktop environment, Full-Featured) on raspiberry pi 4 
and i followed the vnc tutorial but it's failed. 
in console shows:

**root@raspbian:~# apt install ./VNC-Server-6.6.0-Linux-ARM.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'realvnc-vnc-server:armhf' instead of './VNC-Server-6.6.0-Linux-ARM.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 realvnc-vnc-server:armhf : Depends: libc6:armhf (>= 2.4) but it is not going to be installed
                            Depends: libgcc1:armhf (>= 1:4.4.0) but it is not going to be installed
                            Depends: libx11-6:armhf but it is not going to be installed
                            Depends: libxext6:armhf but it is not going to be installed
                            Depends: libxtst6:armhf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.**

once i finished the system installation. i only did: create root account and apt-get update/apt-get upgrade.

so can you guys help me,pls ;)

aiminickwong, 2019-10-17:
@Jyurineko 

You missed the important steps, Maybe you had used the past version readme.

I'm very sorry for you , it's our duty, we forget write some steps, but we had update readme, it will be OK now.

So check the newest version readme, or do a quickly step blow:

```
dpkg --add-architecture armhf ; apt update
cd ~/ ; \
wget https://www.realvnc.com/download/file/vnc.files/VNC-Server-6.6.0-Linux-ARM.deb
apt install ./VNC-Server-6.6.0-Linux-ARM.deb
rm -rf ./VNC-Server-6.6.0-Linux-ARM.deb
dpkg --remove-architecture armhf ; apt update
systemctl enable vncserver-x11-serviced.service ; \
systemctl start vncserver-x11-serviced.service
```
Thank for your eedback and attention. :)

Jyurineko, 2019-10-17:
@aiminickwong 

i followed the latest steps in tutorial and i try your reply steps again. and it shows also:

`root@raspbian:~# dpkg --add-architecture armhf ; apt update
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/debian buster InRelease
Get:2 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates InRelease [39.1 kB]
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports InRelease                                                                                            
Get:5 http://mirrors.tuna.tsinghua.edu.cn/debian buster/main armhf Packages [7,690 kB]                                                                                 
Get:6 http://mirrors.tuna.tsinghua.edu.cn/debian buster/non-free armhf Packages [62.0 kB]                                                                              
Get:7 http://mirrors.tuna.tsinghua.edu.cn/debian buster/contrib armhf Packages [40.9 kB]                                                                               
Get:8 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main armhf Packages [98.1 kB]                                                                 
Get:9 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates/main armhf Packages [5,780 B]                                                                          
Get:10 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main armhf Packages [116 kB]                                                                        
Get:11 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/non-free armhf Packages [5,192 B]                                                                   
Get:12 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/contrib armhf Packages [5,040 B]                                                                    
Fetched 8,062 kB in 15s (545 kB/s)                                                                                                                                     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
root@raspbian:~# cd ~/ ; \
> wget https://www.realvnc.com/download/file/vnc.files/VNC-Server-6.6.0-Linux-ARM.deb
--2019-10-18 01:34:16--  https://www.realvnc.com/download/file/vnc.files/VNC-Server-6.6.0-Linux-ARM.deb
Resolving www.realvnc.com (www.realvnc.com)... 2600:9000:20e8:8600:9:5d24:c640:93a1, 2600:9000:20e8:6e00:9:5d24:c640:93a1, 2600:9000:20e8:3e00:9:5d24:c640:93a1, ...
Connecting to www.realvnc.com (www.realvnc.com)|2600:9000:20e8:8600:9:5d24:c640:93a1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6895442 (6.6M) [application/x-debian-package]
Saving to: ‘VNC-Server-6.6.0-Linux-ARM.deb’

VNC-Server-6.6.0-Linux-ARM.deb            100%[=====================================================================================>]   6.58M  17.2MB/s    in 0.4s    

2019-10-18 01:34:17 (17.2 MB/s) - ‘VNC-Server-6.6.0-Linux-ARM.deb’ saved [6895442/6895442]

root@raspbian:~# apt install ./VNC-Server-6.6.0-Linux-ARM.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'realvnc-vnc-server:armhf' instead of './VNC-Server-6.6.0-Linux-ARM.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 realvnc-vnc-server:armhf : Depends: libc6:armhf (>= 2.4) but it is not going to be installed
                            Depends: libgcc1:armhf (>= 1:4.4.0) but it is not going to be installed
                            Depends: libx11-6:armhf but it is not going to be installed
                            Depends: libxext6:armhf but it is not going to be installed
                            Depends: libxtst6:armhf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`

it doesn't work. :(

ps: i have already give the star, and followed bilibili and bbs~ :) supporting!

Jyurineko, 2019-10-17:
@aiminickwong

i followed the latest steps in tutorial and i try your reply steps again. and it shows also:

root@raspbian:# dpkg --add-architecture armhf ; apt update
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/debian buster InRelease
Get:2 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates InRelease [39.1 kB]
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports InRelease
Get:5 http://mirrors.tuna.tsinghua.edu.cn/debian buster/main armhf Packages [7,690 kB]
Get:6 http://mirrors.tuna.tsinghua.edu.cn/debian buster/non-free armhf Packages [62.0 kB]
Get:7 http://mirrors.tuna.tsinghua.edu.cn/debian buster/contrib armhf Packages [40.9 kB]
Get:8 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main armhf Packages [98.1 kB]
Get:9 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates/main armhf Packages [5,780 B]
Get:10 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main armhf Packages [116 kB]
Get:11 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/non-free armhf Packages [5,192 B]
Get:12 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/contrib armhf Packages [5,040 B]
Fetched 8,062 kB in 15s (545 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@raspbian:# cd ~/ ; \

wget https://www.realvnc.com/download/file/vnc.files/VNC-Server-6.6.0-Linux-ARM.deb
--2019-10-18 01:34:16-- https://www.realvnc.com/download/file/vnc.files/VNC-Server-6.6.0-Linux-ARM.deb
Resolving www.realvnc.com (www.realvnc.com)... 2600:9000:20e8:8600:9:5d24:c640:93a1, 2600:9000:20e8:6e00:9:5d24:c640:93a1, 2600:9000:20e8:3e00:9:5d24:c640:93a1, ...
Connecting to www.realvnc.com (www.realvnc.com)|2600:9000:20e8:8600:9:5d24:c640:93a1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6895442 (6.6M) [application/x-debian-package]
Saving to: ‘VNC-Server-6.6.0-Linux-ARM.deb’

VNC-Server-6.6.0-Linux-ARM.deb 100%[=====================================================================================>] 6.58M 17.2MB/s in 0.4s

2019-10-18 01:34:17 (17.2 MB/s) - ‘VNC-Server-6.6.0-Linux-ARM.deb’ saved [6895442/6895442]

root@raspbian:~# apt install ./VNC-Server-6.6.0-Linux-ARM.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'realvnc-vnc-server:armhf' instead of './VNC-Server-6.6.0-Linux-ARM.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
realvnc-vnc-server:armhf : Depends: libc6:armhf (>= 2.4) but it is not going to be installed
Depends: libgcc1:armhf (>= 1:4.4.0) but it is not going to be installed
Depends: libx11-6:armhf but it is not going to be installed
Depends: libxext6:armhf but it is not going to be installed
Depends: libxtst6:armhf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

it doesn't work. :(

ps: i have already give the star, and followed bilibili and bbs~ :) supporting!

aiminickwong, 2019-10-18:
It's already show a message:  *Depends: libc6:armhf (>= 2.4) but it is not going to be installed*
so just try install it:

```
apt install libc6:armhf -y
```

@Jyurineko

aiminickwong, 2019-10-18:
We had tested in newest version again, No problems anymore.

Jyurineko, 2019-10-18:
@aiminickwong 
i reinstall the system "2019-10-13-OPENFANS-Debian-Buster-Desktop-Aarch64-ext4-v2019-2.0-Release" this version.
after three times reboot. in web console i type the code.
______________________________
**first try:**

pi@raspbian:~$ sudo apt-get update
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/debian buster InRelease        
Get:2 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates InRelease [39.1 kB]
Get:3 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates InRelease [49.3 kB]
Get:4 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports InRelease [46.7 kB]
Get:5 http://mirrors.tuna.tsinghua.edu.cn/debian buster/main Translation-zh [1,526 B]                                                                                  
Get:6 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main arm64 Packages [98.3 kB]                                                                 
Get:7 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main Translation-en [61.4 kB]                                                                 
Get:8 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages.diff/Index [27.8 kB]                                                             
Get:9 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main Translation-en.diff/Index [27.8 kB]                                                             
Get:10 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-13-2014.06.pdiff [874 B]                                                
Get:11 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-14-1416.13.pdiff [6,648 B]                                              
Get:12 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-14-2013.06.pdiff [6,198 B]                                              
Get:13 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-17-2015.25.pdiff [186 B]                                                
Get:14 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-18-0215.12.pdiff [184 B]                                                
Get:14 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main arm64 Packages 2019-10-18-0215.12.pdiff [184 B]                                                
Get:15 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main Translation-en 2019-10-14-1416.13.pdiff [9,122 B]                                              
Get:16 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main Translation-en 2019-10-14-2013.06.pdiff [999 B]                                                
Get:17 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main Translation-en 2019-10-16-2015.16.pdiff [241 B]                                                
Get:17 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports/main Translation-en 2019-10-16-2015.16.pdiff [241 B]                                                
Fetched 376 kB in 9s (43.0 kB/s)                                                                                                                                       
Reading package lists... Done
pi@raspbian:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libllvm7
Use 'sudo apt autoremove' to remove it.
The following packages will be upgraded:
  libssl1.1 libunbound8 openssl sudo
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,843 kB of archives.
After this operation, 9,216 B disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main arm64 libssl1.1 arm64 1.1.1d-0+deb10u2 [1,381 kB]
Get:2 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main arm64 libunbound8 arm64 1.9.0-2+deb10u1 [432 kB]
Get:3 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main arm64 openssl arm64 1.1.1d-0+deb10u2 [823 kB]
Get:4 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates/main arm64 sudo arm64 1.8.27-1+deb10u1 [1,207 kB]
Fetched 3,843 kB in 5s (728 kB/s)
Preconfiguring packages ...
(Reading database ... 133377 files and directories currently installed.)
Preparing to unpack .../libssl1.1_1.1.1d-0+deb10u2_arm64.deb ...
Unpacking libssl1.1:arm64 (1.1.1d-0+deb10u2) over (1.1.1d-0+deb10u1) ...
Preparing to unpack .../libunbound8_1.9.0-2+deb10u1_arm64.deb ...
Unpacking libunbound8:arm64 (1.9.0-2+deb10u1) over (1.9.0-2) ...
Preparing to unpack .../openssl_1.1.1d-0+deb10u2_arm64.deb ...
Unpacking openssl (1.1.1d-0+deb10u2) over (1.1.1d-0+deb10u1) ...
Preparing to unpack .../sudo_1.8.27-1+deb10u1_arm64.deb ...
Unpacking sudo (1.8.27-1+deb10u1) over (1.8.27-1) ...
Setting up libssl1.1:arm64 (1.1.1d-0+deb10u2) ...
Setting up libunbound8:arm64 (1.9.0-2+deb10u1) ...
Setting up sudo (1.8.27-1+deb10u1) ...
Setting up openssl (1.1.1d-0+deb10u2) ...
Processing triggers for systemd (241-7~deb10u1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.29-2) ...
pi@raspbian:~$ sudo apt install libc6:armhf -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
[... cut ...]
_______________________________________________________________________________________
[... 137 lines cut ...]
then i try to use root account to do it again.
also the same.

i didn't know what'going on.

我尝试了三次 
步骤说明：
重新装最新的10-13的桌面完全版那个系统。
自动重启稳定了之后 我用chrome ip地址:9090 进console。
然后先apt-get update /upgrade
再直接apt install libc6:armhf -y
意识到是要32位先启动。

所以尝试了您的建议 
dpkg --add-architecture armhf ; apt update
apt install libc6:armhf -y

还是不行。

然后我开root账户 再试了一遍 还是不行。
请问是源的问题吗还是什么问题 
我是google drive那儿下载的系统镜像
下载好烧完系统 什么都没动 直接console 打代码的

不好意思啊 这个问题打扰这么久。

aiminickwong, 2019-10-18:
No problem :）
from your step, I found you miss this step

```
dpkg --add-architecture armhf ; apt update 
```

Jyurineko, 2019-10-18:
so your advice is 
1. dpkg --add-architecture armhf
2. apt update
3. apt install libc6:armhf -y

right? i will try it again.

Jyurineko, 2019-10-18:
it really doesn't work :(

Jyurineko, 2019-10-18:
@aiminickwong  i think you didn't see my second try and third try.
the first try i do not type "dpkg --add-architecture armhf ; apt update "
but at second/third i do type firstly "dpkg --add-architecture armhf ; apt update ".

aiminickwong, 2019-10-18:
solved.

closed 2019-10-18 by aiminickwong

gitOtonaut, 2020-06-05:
hi, i tried to get realvnc running on pi4 ubuntu19.10 64bit aarch64 userland half a year before, so adding armhf architecture did not help @ all, can someone tell me how you solved this issue, thanks in advance :-)

TURX, 2020-06-16:
> hi, i tried to get realvnc running on pi4 ubuntu19.10 64bit aarch64 userland half a year before, so adding armhf architecture did not help @ all, can someone tell me how you solved this issue, thanks in advance :-)

Same to me. I am using Ubuntu 20.04 aarch64 on Pi3B.

gitOtonaut, 2020-08-23:
So, as i have spent much time on raspbian the last 6months i return to ubuntu and upgrade from 19.10 to 20.04, if some one had any success in installing realvnc server on raspberry with ubuntu please let me know :-)

wh054, 2020-08-27:
The same problem as me

gitOtonaut, 2020-08-27:
ohhh WTApainITA, so i gave up on 64bit realvnc-server, i have successfully installed my whole AstroImaging control Software Kstars/Ekos on Ubuntu20.04 armhf with realvnc-server remote controled which works like charm, its such a nice thing to have a full working Ubuntu20.04 on a PI4 ;-P