openfans-community-offical/Debian-Pi-Aarch64 #53: xrdp can't work

walkonbothsides, 2019-10-18:
[image]
I installed tightvncserver and xrdp,but it can't work.
This is xrdp.log:
`[20191019-01:49:31] [INFO ] starting xrdp with pid 27675
[20191019-01:49:31] [INFO ] listening to port 3389 on 0.0.0.0
[20191019-01:49:36] [INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.1.130 port 54362
[20191019-01:49:36] [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.87 port 3389)
[20191019-01:49:36] [DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
[20191019-01:49:36] [INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
[20191019-01:49:36] [INFO ] Using default X.509 key file: /etc/xrdp/key.pem
[20191019-01:49:36] [DEBUG] TLSv1.3 enabled
[20191019-01:49:36] [DEBUG] TLSv1.2 enabled
[20191019-01:49:36] [DEBUG] Security layer: requested 11, selected 1
[20191019-01:49:36] [INFO ] connected client computer name: MEGAFLOPS-PC
[20191019-01:49:36] [INFO ] adding channel item name rdpdr chan_id 1004 flags 0x80800000
[20191019-01:49:36] [INFO ] adding channel item name rdpsnd chan_id 1005 flags 0xc0000000
[20191019-01:49:36] [INFO ] adding channel item name cliprdr chan_id 1006 flags 0xc0a00000
[20191019-01:49:36] [INFO ] adding channel item name drdynvc chan_id 1007 flags 0xc0800000
[20191019-01:49:36] [INFO ] TLS connection established from ::ffff:192.168.1.130 port 54362: TLSv1.2 with cipher ECDHE-RSA-AES256-GCM-SHA384
[20191019-01:49:36] [DEBUG] xrdp_00006c64_wm_login_mode_event_00000001
[20191019-01:49:36] [INFO ] Cannot find keymap file /etc/xrdp/km-00000804.ini
[20191019-01:49:36] [INFO ] Cannot find keymap file /etc/xrdp/km-00000804.ini
[20191019-01:49:36] [INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
[20191019-01:49:36] [WARN ] local keymap file for 0x00000804 found and doesn't match built in keymap, using local keymap file
[20191019-01:49:41] [DEBUG] xrdp_wm_log_msg: connecting to sesman ip 127.0.0.1 port 3350
[20191019-01:49:42] [INFO ] xrdp_wm_log_msg: sesman connect ok
[20191019-01:49:42] [DEBUG] xrdp_wm_log_msg: sending login info to session manager, please wait...
[20191019-01:49:42] [DEBUG] return value from xrdp_mm_connect 0
[20191019-01:49:42] [INFO ] xrdp_wm_log_msg: login successful for display 10
[20191019-01:49:42] [DEBUG] xrdp_wm_log_msg: started connecting
[20191019-01:49:43] [INFO ] lib_mod_log_peer: xrdp_pid=27748 connected to X11rdp_pid=27841 X11rdp_uid=1000 X11rdp_gid=27 client_ip=::ffff:192.168.1.130 client_port=54362
[20191019-01:49:43] [DEBUG] xrdp_wm_log_msg: connected ok
[20191019-01:49:43] [DEBUG] xrdp_mm_connect_chansrv: chansrv connect successful
[20191019-01:49:43] [DEBUG] Closed socket 18 (AF_INET6 ::1 port 35938)
[20191019-01:49:43] [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.87 port 3389)
[20191019-01:49:43] [DEBUG] xrdp_mm_module_cleanup
[20191019-01:49:43] [DEBUG] Closed socket 19 (AF_UNIX)
[20191019-01:49:43] [DEBUG] Closed socket 20 (AF_UNIX)
[20191019-01:54:19] [INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.1.130 port 54706
[20191019-01:54:19] [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.87 port 3389)
[20191019-01:54:19] [DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
[20191019-01:54:19] [INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
[20191019-01:54:19] [INFO ] Using default X.509 key file: /etc/xrdp/key.pem
[20191019-01:54:19] [DEBUG] TLSv1.3 enabled
[20191019-01:54:19] [DEBUG] TLSv1.2 enabled
[20191019-01:54:19] [DEBUG] Security layer: requested 11, selected 1
[20191019-01:54:19] [INFO ] connected client computer name: MEGAFLOPS-PC
[20191019-01:54:19] [INFO ] adding channel item name rdpdr chan_id 1004 flags 0x80800000
[20191019-01:54:19] [INFO ] adding channel item name rdpsnd chan_id 1005 flags 0xc0000000
[20191019-01:54:19] [INFO ] adding channel item name cliprdr chan_id 1006 flags 0xc0a00000
[20191019-01:54:19] [INFO ] adding channel item name drdynvc chan_id 1007 flags 0xc0800000
[20191019-01:54:19] [INFO ] TLS connection established from ::ffff:192.168.1.130 port 54706: TLSv1.2 with cipher ECDHE-RSA-AES256-GCM-SHA384
[20191019-01:54:19] [DEBUG] xrdp_00007a4a_wm_login_mode_event_00000001
[20191019-01:54:19] [INFO ] Cannot find keymap file /etc/xrdp/km-00000804.ini
[20191019-01:54:19] [INFO ] Cannot find keymap file /etc/xrdp/km-00000804.ini
[20191019-01:54:19] [INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
[20191019-01:54:19] [WARN ] local keymap file for 0x00000804 found and doesn't match built in keymap, using local keymap file
[20191019-01:54:23] [DEBUG] xrdp_wm_log_msg: connecting to sesman ip 127.0.0.1 port 3350
[20191019-01:54:24] [INFO ] xrdp_wm_log_msg: sesman connect ok
[20191019-01:54:24] [DEBUG] xrdp_wm_log_msg: sending login info to session manager, please wait...
[20191019-01:54:24] [DEBUG] return value from xrdp_mm_connect 0
[20191019-01:54:24] [INFO ] xrdp_wm_log_msg: login successful for display 10
[20191019-01:54:24] [DEBUG] xrdp_wm_log_msg: started connecting
[20191019-01:54:24] [INFO ] lib_mod_log_peer: xrdp_pid=31306 connected to X11rdp_pid=31374 X11rdp_uid=1000 X11rdp_gid=27 client_ip=::ffff:192.168.1.130 client_port=54706
[20191019-01:54:24] [DEBUG] xrdp_wm_log_msg: connected ok
[20191019-01:54:24] [DEBUG] xrdp_mm_connect_chansrv: chansrv connect successful
[20191019-01:54:24] [DEBUG] Closed socket 18 (AF_INET6 ::1 port 35944)
[20191019-01:54:25] [DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.1.87 port 3389)
[20191019-01:54:25] [DEBUG] xrdp_mm_module_cleanup
[20191019-01:54:25] [DEBUG] Closed socket 19 (AF_UNIX)
[... 41 lines cut ...]
[20191019-01:49:43] [DEBUG] cleanup_sockets:
[20191019-01:49:43] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdp_chansrv_audio_out_socket_10
[20191019-01:49:43] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdp_chansrv_audio_in_socket_10
[20191019-01:49:43] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdpapi_10
[20191019-01:49:43] [INFO ] ++ terminated session:  username pi, display :10.0, session_pid 27837, ip ::ffff:192.168.1.130:54362 - socket: 12
[20191019-01:54:23] [INFO ] A connection received from ::1 port 35944
[20191019-01:54:24] [INFO ] ++ created session (access granted): username pi, ip ::ffff:192.168.1.130:54706 - socket: 12
[20191019-01:54:24] [INFO ] starting Xorg session...
[20191019-01:54:24] [DEBUG] Closed socket 9 (AF_INET6 :: port 5910)
[20191019-01:54:24] [DEBUG] Closed socket 9 (AF_INET6 :: port 6010)
[20191019-01:54:24] [DEBUG] Closed socket 9 (AF_INET6 :: port 6210)
[20191019-01:54:24] [DEBUG] Closed socket 8 (AF_INET6 ::1 port 3350)
[20191019-01:54:24] [INFO ] calling auth_start_session from pid 31367
[20191019-01:54:24] [DEBUG] Closed socket 7 (AF_INET6 ::1 port 3350)
[20191019-01:54:24] [DEBUG] Closed socket 8 (AF_INET6 ::1 port 3350)
[20191019-01:54:24] [INFO ] /usr/lib/xorg/Xorg :10 -auth .Xauthority -config xrdp/xorg.conf -noreset -nolisten tcp -logfile .xorgxrdp.%s.log  
[20191019-01:54:24] [CORE ] waiting for window manager (pid 31373) to exit
[20191019-01:54:24] [CORE ] window manager (pid 31373) did exit, cleaning up session
[20191019-01:54:25] [INFO ] calling auth_stop_session and auth_end from pid 31367
[20191019-01:54:25] [DEBUG] cleanup_sockets:
[20191019-01:54:25] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdp_chansrv_audio_out_socket_10
[20191019-01:54:25] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdp_chansrv_audio_in_socket_10
[20191019-01:54:25] [DEBUG] cleanup_sockets: deleting /var/run/xrdp/sockdir/xrdpapi_10
[20191019-01:54:25] [INFO ] ++ terminated session:  username pi, display :10.0, session_pid 31367, ip ::ffff:192.168.1.130:54706 - socket: 12`
How to solve it?I used the `xfce` version

aiminickwong, 2019-10-18:
@walkonbothsides tightvncserver and xrdp with a known issue in 64-bit mode by tightvncserver‘s own problem, we suggest you use realvnc with reference to our documentation.

closed 2019-10-18 by aiminickwong

walkonbothsides, 2019-10-19:
OK Thanks.But nothing serious now,My RPi 4 was burned several minutes ago.