openfans-community-offical/Debian-Pi-Aarch64 #54: VNC can't installed

yaoyiqun, 2019-10-19:
pi@raspbian:~$ ls
VNC-Server-6.6.0-Linux-ARM.deb  Wallpapers  下载  公共  图片  文档  桌面  模板  视频  音乐
pi@raspbian:~$ sudo dpkg --add-architecture armhf
pi@raspbian:~$ sudo apt update
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/debian buster InRelease
Hit:2 http://mirrors.tuna.tsinghua.edu.cn/debian-security buster/updates InRelease
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/debian buster-updates InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/debian buster-backports InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
pi@raspbian:~$ sudo apt install ./VNC-Server-6.6.0-Linux-ARM.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'realvnc-vnc-server:armhf' instead of './VNC-Server-6.6.0-Linux-ARM.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 realvnc-vnc-server:armhf : Depends: libc6:armhf (>= 2.4) but it is not going to be installed
                            Depends: libgcc1:armhf (>= 1:4.4.0) but it is not going to be installed
                            Depends: libx11-6:armhf but it is not going to be installed
                            Depends: libxext6:armhf but it is not going to be installed
                            Depends: libxtst6:armhf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
pi@raspbian:~$ sudo apt install libc6:armhf
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ffmpeg : Depends: libc6 (>= 2.29) but it is not going to be installed
          Depends: libsdl2-2.0-0 (>= 2.0.10) but it is not going to be installed
 libavcodec58 : Depends: libaom0 (>= 1.0.0) but it is not going to be installed
                Depends: libc6 (>= 2.29) but it is not going to be installed
                Depends: libcairo2 (>= 1.2.4) but it is not going to be installed
                Depends: libcodec2-0.8.1 but it is not going to be installed
                Depends: libglib2.0-0 (>= 2.12.0) but it is not going to be installed
                Depends: libgsm1 (>= 1.0.18) but it is not going to be installed
                Depends: liblzma5 (>= 5.1.1alpha+20120614) but it is not going to be installed
                Depends: libmp3lame0 (>= 3.100) but it is not going to be installed
                Depends: libopenjp2-7 (>= 2.0.0) but it is not going to be installed
                Depends: libopus0 (>= 1.1) but it is not going to be installed
                Depends: librsvg2-2 (>= 2.14.4) but it is not going to be installed
                Depends: libshine3 (>= 3.1.0) but it is not going to be installed
                Depends: libsnappy1v5 but it is not going to be installed
                Depends: libspeex1 (>= 1.2~beta3-1) but it is not going to be installed
                Depends: libtheora0 (>= 1.0) but it is not going to be installed
                Depends: libtwolame0 (>= 0.3.10) but it is not going to be installed
                Depends: libva2 (>= 2.2.0) but it is not going to be installed
                Depends: libvorbis0a (>= 1.1.2) but it is not going to be installed
                Depends: libvorbisenc2 (>= 1.1.2) but it is not going to be installed
                Depends: libvpx6 (>= 1.6.0) but it is not going to be installed
                Depends: libwavpack1 (>= 4.40.0) but it is not going to be installed
                Depends: libwebp6 (>= 0.5.1) but it is not going to be installed
                Depends: libwebpmux3 (>= 0.6.1-2+b1) but it is not going to be installed
                Depends: libx264-155 but it is not going to be installed
                Depends: libx265-176 (>= 3.0) but it is not going to be installed
                Depends: libxvidcore4 (>= 1.2.2) but it is not going to be installed
                Depends: libzvbi0 (>= 0.2.35) but it is not going to be installed
[... 37 lines cut ...]
                Depends: libvidstab1.1 but it is not going to be installed
                Depends: libzmq5 (>= 3.2.3+dfsg) but it is not going to be installed
 libavformat58 : Depends: libbluray2 (>= 1:0.2.2) but it is not going to be installed
                 Depends: libbz2-1.0 but it is not going to be installed
                 Depends: libc6 (>= 2.28) but it is not going to be installed
                 Depends: libchromaprint1 (>= 1.3.2) but it is not going to be installed
                 Depends: libgme0 (>= 0.5.5) but it is not going to be installed
                 Depends: libgnutls30 (>= 3.6.6) but it is not going to be installed
                 Depends: libopenmpt0 (>= 0.3.0) but it is not going to be installed
                 Depends: libssh-gcrypt-4 (>= 0.8.0) but it is not going to be installed
                 Depends: libxml2 (>= 2.7.4) but it is not going to be installed
                 Depends: zlib1g (>= 1:1.2.0.2) but it is not going to be installed
 libavresample4 : Depends: libc6 (>= 2.17) but it is not going to be installed
 libavutil56 : Depends: libc6 (>= 2.29) but it is not going to be installed
               Depends: libdrm2 (>= 2.3.1) but it is not going to be installed
               Depends: libva-drm2 (>= 1.3) but it is not going to be installed
               Depends: libva-x11-2 (>= 1.3) but it is not going to be installed
               Depends: libva2 (>= 2.1.0) but it is not going to be installed
               Depends: libvdpau1 (>= 0.2) but it is not going to be installed
               Depends: libx11-6 but it is not going to be installed
 libpostproc55 : Depends: libc6 (>= 2.17) but it is not going to be installed
 libswresample3 : Depends: libc6 (>= 2.29) but it is not going to be installed
                  Depends: libsoxr0 (>= 0.1.0) but it is not going to be installed
 libswscale5 : Depends: libc6 (>= 2.29) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

yaoyiqun, 2019-10-19:
my device is raspberry 3b

aiminickwong, 2019-10-19:
@yaoyiqun the same issue #51, has beed soved, Please check the newest readme. [Here](https://github.com/openfans-community-offical/Debian-Pi-Aarch64/blob/master/README.md#3-10-install-vnc-remote-desktop-of-macos-mojave-theme-desktop-environment)

closed 2019-10-19 by aiminickwong