openfans-community-offical/Debian-Pi-Aarch64 #62: RaspberryPi 4 no HDMI audio

SuLingGG, 2019-11-23:
My RaspberryPi is 2G Ram version, 
After a new installation image, I use hdmi0 connect to TV by HDMI cable.
I use VLC to play music, but it can't output audio correctly. 
In most time it is mute, but it occasionally plays out some noise,
I'm sure it  is not caused by config.txt because I have a new installation.
But when I use raspbian, the HDMI audio works fine.
Hope to fix it.

aiminickwong, 2019-11-23:
What about your image version, why you make sure it's not problem by config.txt cuz you have a new installation? So, upload more log and msg.

SuLingGG, 2019-11-25:
> What about your image version, why you make sure it's not problem by config.txt cuz you have a new installation? So, upload more log and msg.

I use this image: `2019-11-17-OPENFANS-Debian-Buster-Desktop-Aarch64-ext4-v2019-2.0-Release.img.xz`

When I found this problem, I flash this image to a new SD card again, then I don't modified any files include config.txt then choose hdmi0 interface and use HDMI cable to connect TV, every thing is ok but no audio found.

So I am sure this situation is not caused by config.txt or any files because I havn't modified any files.

I flash the same image to another SD card, The situation comes again.

I "Test" the zip file in Bandizip, it shows "No error found in this file", so the file is not broken on my computer.

But I use the same SD card to flash official raspbian, use same HDMI interface/the same cable connect to the same TV (TV's HDMI interface is same, too) to test again, the audio works well.

So I think this problem is caused by this image.

Logs will be upload about 2~3 hours, I will flash the same image file and don't modify any files again to catch logs.

SuLingGG, 2019-11-25:
This zip archive include my config.txt and /var/log directory.
[logs.zip](https://github.com/openfans-community-offical/Debian-Pi-Aarch64/files/3886989/logs.zip)

aiminickwong, 2019-11-25:
According to the logs provided, we have not found any problems, but this may be a known compatibility issue. 

According to different versions of the kernel or firmware, you can refer to:

https://discourse.osmc.tv/t/raspberry-pi-3-new-install-no-hdmi-audio/16956/9

https://www.raspberrypi.org/forums/viewtopic.php?t=248094

closed 2019-11-29 by aiminickwong