openfans-community-offical/Debian-Pi-Aarch64 #64: docker不能正常工作

batcom, 2019-11-29:
我已经安装了最下版本到pi3+上，但是执行docker info 报
SIGILL: illegal instruction 错误.docker不能正常工作

uname -a
Linux raspbian 4.19.85-OPENFANS+20191125-v8 #1 SMP PREEMPT Mon Nov 25 16:40:06 CST 2019 aarch64 GNU/Linux

docker info
SIGILL: illegal instruction
PC=0x5562dae168 m=0 sigcode=1

goroutine 1 [running, locked to thread]:
strings.(*Builder).copyCheck(...)
	/usr/local/go/src/strings/builder.go:39
strings.(*Builder).WriteRune(0x40000ebd88, 0x75, 0x40000f3428, 0x0, 0x40000ebda8)
	/usr/local/go/src/strings/builder.go:104 +0x240 fp=0x40000ebd50 sp=0x40000ebce0 pc=0x5562dae168
regexp/syntax.(*Prog).Prefix(0x40000b8a80, 0x4000104360, 0x0, 0x0)
	/usr/local/go/src/regexp/syntax/prog.go:158 +0x8c fp=0x40000ebdb0 sp=0x40000ebd50 pc=0x5562df1c94
regexp.compile(0x5564004b16, 0x1a, 0x40000000d4, 0x5562e66ea0, 0x40000ebee8, 0x32257e1)
	/usr/local/go/src/regexp/regexp.go:195 +0x2a0 fp=0x40000ebe40 sp=0x40000ebdb0 pc=0x5562dfe318
regexp.Compile(...)
	/usr/local/go/src/regexp/regexp.go:131
regexp.MustCompile(0x5564004b16, 0x1a, 0x5563fe1ec8)
	/usr/local/go/src/regexp/regexp.go:270 +0x40 fp=0x40000ebed0 sp=0x40000ebe40 pc=0x5562dfe9b8
github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools.init.ializers()
	/go/src/github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools/usergroupadd_linux.go:27 +0x148 fp=0x40000ebf10 sp=0x40000ebed0 pc=0x5562e66ef0
github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools.init()
	<autogenerated>:1 +0x90 fp=0x40000ebf20 sp=0x40000ebf10 pc=0x5562e67008
github.com/docker/cli/vendor/github.com/docker/docker/pkg/homedir.init()
	<autogenerated>:1 +0x60 fp=0x40000ebf30 sp=0x40000ebf20 pc=0x5562e67278
github.com/docker/cli/cli/config.init()
	<autogenerated>:1 +0x70 fp=0x40000ebf40 sp=0x40000ebf30 pc=0x5562e67e08
github.com/docker/cli/cli/command.init()
	<autogenerated>:1 +0x70 fp=0x40000ebf50 sp=0x40000ebf40 pc=0x5563487d68
github.com/docker/cli/cli-plugins/manager.init()
	<autogenerated>:1 +0x58 fp=0x40000ebf60 sp=0x40000ebf50 pc=0x556348ae90
github.com/docker/cli/cli.init()
	<autogenerated>:1 +0x60 fp=0x40000ebf70 sp=0x40000ebf60 pc=0x556348ebb8
main.init()
	<autogenerated>:1 +0x68 fp=0x40000ebf80 sp=0x40000ebf70 pc=0x5563fd7d20
runtime.main()
	/usr/local/go/src/runtime/proc.go:188 +0x1d8 fp=0x40000ebfd0 sp=0x40000ebf80 pc=0x5562cfc8b0
runtime.goexit()
	/usr/local/go/src/runtime/asm_arm64.s:1128 +0x4 fp=0x40000ebfd0 sp=0x40000ebfd0 pc=0x5562d275dc

r0      0x0
r1      0x40000ea370
r2      0x40000ebd50
r3      0x40000ebd88
r4      0x0
r5      0x0
r6      0x60
r7      0x1000
r8      0xc
r9      0x55642d94dc
r10     0x40001043ef
r11     0x1000
r12     0x1
r13     0x55642d94da
r14     0x0
r15     0xfa
r16     0x0
r17     0x0
r18     0x11f
r19     0x8
r20     0x18
r21     0x5565dcbe20
r22     0x5565de92d4
r23     0x5565de92d5
r24     0x5565dc7620
r25     0x18
r26     0x0
r27     0xffffffffffffffcc
r28     0x4000000180
r29     0x40000ebcd8
lr      0x5562df1c94
sp      0x40000ebce0
pc      0x5562dae168
fault   0x0

aiminickwong, 2019-11-29:
Check you system img md5 and shasum. **That's not our system problem.** Issue will be closed.
```
pi@raspbian:~$ dmesg|grep 'Machine model';uname -r;sudo systemctl status docker
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
4.19.85-OPENFANS+20191125-v8
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-30 00:09:16 CST; 4min 1s ago
     Docs: https://docs.docker.com
 Main PID: 2251 (dockerd)
    Tasks: 14
   Memory: 42.9M
   CGroup: /system.slice/docker.service
           └─2251 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock

Nov 30 00:09:12 raspbian dockerd[2251]: time="2019-11-30T00:09:12.491042783+08:00" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd
Nov 30 00:09:12 raspbian dockerd[2251]: time="2019-11-30T00:09:12.491118148+08:00" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Nov 30 00:09:12 raspbian dockerd[2251]: time="2019-11-30T00:09:12.636307887+08:00" level=info msg="[graphdriver] using prior storage driver: overlay2"
Nov 30 00:09:12 raspbian dockerd[2251]: time="2019-11-30T00:09:12.715268981+08:00" level=info msg="Loading containers: start."
Nov 30 00:09:14 raspbian dockerd[2251]: time="2019-11-30T00:09:14.674772314+08:00" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon
Nov 30 00:09:15 raspbian dockerd[2251]: time="2019-11-30T00:09:15.104600647+08:00" level=info msg="Loading containers: done."
Nov 30 00:09:16 raspbian dockerd[2251]: time="2019-11-30T00:09:16.047363876+08:00" level=info msg="Docker daemon" commit=a872fc2 graphdriver(s)=overlay2 version=19.03.3
Nov 30 00:09:16 raspbian dockerd[2251]: time="2019-11-30T00:09:16.065727365+08:00" level=info msg="Daemon has completed initialization"
Nov 30 00:09:16 raspbian dockerd[2251]: time="2019-11-30T00:09:16.394597782+08:00" level=info msg="API listen on /var/run/docker.sock"
Nov 30 00:09:16 raspbian systemd[1]: Started Docker Application Container Engine.
lines 1-20/20 (END)
```

And **docker info** :

```
docker info
Client:
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 19.03.3
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 894b81a4b802e4eb2a91d1ce216b8817763c29fb
 runc version: 425e105d5a03fabd737a126ad93d62a9eeede87f
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.19.85-OPENFANS+20191125-v8
 Operating System: Debian GNU/Linux 10 (buster)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 733.4MiB
 Name: raspbian
 ID: PKNZ:KXJX:FOK5:22KV:B426:22IB:PIFS:IBD7:MX6V:JSWR:DUAD:TVYD
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```

batcom, 2019-11-30:
md5sum 2019-11-25-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-Release.img 
68672c4d83d4fcc82fb5bcd9e3e7e143  2019-11-25-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-Release.img

I have checked the img md5 ,it is equal the value on baidunetdisk .

`pi@raspbian:~$ dmesg|grep 'Machine model';uname -r;sudo systemctl status docker
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
4.19.85-OPENFANS+20191125-v8
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: https://docs.docker.com
`
and docker info
`pi@raspbian:~$ docker info
SIGILL: illegal instruction
PC=0x5577640168 m=0 sigcode=1

goroutine 1 [running, locked to thread]:
strings.(*Builder).copyCheck(...)
	/usr/local/go/src/strings/builder.go:39
strings.(*Builder).WriteRune(0x40000ebd88, 0x75, 0x40000f3428, 0x0, 0x40000ebda8)
	/usr/local/go/src/strings/builder.go:104 +0x240 fp=0x40000ebd50 sp=0x40000ebce0 pc=0x5577640168
regexp/syntax.(*Prog).Prefix(0x40000baa80, 0x4000104360, 0x0, 0x0)
	/usr/local/go/src/regexp/syntax/prog.go:158 +0x8c fp=0x40000ebdb0 sp=0x40000ebd50 pc=0x5577683c94
regexp.compile(0x5578896b16, 0x1a, 0x40000000d4, 0x55776f8ea0, 0x40000ebee8, 0x88fbdbd6)
	/usr/local/go/src/regexp/regexp.go:195 +0x2a0 fp=0x40000ebe40 sp=0x40000ebdb0 pc=0x5577690318
regexp.Compile(...)
	/usr/local/go/src/regexp/regexp.go:131
regexp.MustCompile(0x5578896b16, 0x1a, 0x5578873ec8)
	/usr/local/go/src/regexp/regexp.go:270 +0x40 fp=0x40000ebed0 sp=0x40000ebe40 pc=0x55776909b8
github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools.init.ializers()
	/go/src/github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools/usergroupadd_linux.go:27 +0x148 fp=0x40000ebf10 sp=0x40000ebed0 pc=0x55776f8ef0
github.com/docker/cli/vendor/github.com/docker/docker/pkg/idtools.init()
	<autogenerated>:1 +0x90 fp=0x40000ebf20 sp=0x40000ebf10 pc=0x55776f9008
github.com/docker/cli/vendor/github.com/docker/docker/pkg/homedir.init()
	<autogenerated>:1 +0x60 fp=0x40000ebf30 sp=0x40000ebf20 pc=0x55776f9278
github.com/docker/cli/cli/config.init()
	<autogenerated>:1 +0x70 fp=0x40000ebf40 sp=0x40000ebf30 pc=0x55776f9e08
github.com/docker/cli/cli/command.init()
	<autogenerated>:1 +0x70 fp=0x40000ebf50 sp=0x40000ebf40 pc=0x5577d19d68
github.com/docker/cli/cli-plugins/manager.init()
	<autogenerated>:1 +0x58 fp=0x40000ebf60 sp=0x40000ebf50 pc=0x5577d1ce90
github.com/docker/cli/cli.init()
	<autogenerated>:1 +0x60 fp=0x40000ebf70 sp=0x40000ebf60 pc=0x5577d20bb8
main.init()
	<autogenerated>:1 +0x68 fp=0x40000ebf80 sp=0x40000ebf70 pc=0x5578869d20
runtime.main()
	/usr/local/go/src/runtime/proc.go:188 +0x1d8 fp=0x40000ebfd0 sp=0x40000ebf80 pc=0x557758e8b0
runtime.goexit()
	/usr/local/go/src/runtime/asm_arm64.s:1128 +0x4 fp=0x40000ebfd0 sp=0x40000ebfd0 pc=0x55775b95dc

r0      0x0
r1      0x40000ea370
r2      0x40000ebd50
r3      0x40000ebd88
r4      0x0
r5      0x0
r6      0x60
r7      0x1000
r8      0xc
r9      0x5578b6b4dc
r10     0x40001043ef
r11     0x1000
r12     0x1
r13     0x5578b6b4da
r14     0x0
r15     0xfa
r16     0x0
r17     0x0
r18     0x11f
r19     0x8
r20     0x18
r21     0x557a65de20
r22     0x557a67b2d4
r23     0x557a67b2d5
r24     0x557a659620
r25     0x18
r26     0x0
r27     0xffffffffffffffcc
r28     0x4000000300
r29     0x40000ebcd8
lr      0x5577683c94
sp      0x40000ebce0
pc      0x5577640168
fault   0x0
`
I have try many times ,please help me. thks;

batcom, 2019-11-30:
and you can try reboot your system once time ,and try run docker info will reproduce the issue.
i try many time ,after install docker completely ,everythings is ok(include docker info),but once reboot the system ,the issue is reproduced.
@aiminickwong

aiminickwong, 2019-11-30:
@batcom 

Unfortunately, we performed the same test **more than 3 times on 3B +** (With version **2019-11-25-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-Release.img**),  but we couldn't reproduce your problem at all, and all the functions of Docker are normal,  even restart many times without any error.`

Now I suggest you carefully check if your TF card has an error during the system burning process?

And I see U said " `i try many time ,after install docker completely ,everythings is ok(include docker info),but once reboot the system ,the issue is reproduced.` "

Why you were installing docker ?? Why ? 

**Our system already contains the latest Docker, why should you install it by yourself?**

closed 2019-12-02 by aiminickwong