openfans-community-offical/Debian-Pi-Aarch64 #71: 使用 sudo apt install libc6:armhf 安装32位库支持的时候出现如下错误

JuserZhang, 2019-12-19:
The following packages have unmet dependencies:
 ffmpeg : Depends: libc6 (>= 2.29) but it is not going to be installed
          Depends: libsdl2-2.0-0 (>= 2.0.9) but it is not going to be installed
 libavcodec58 : Depends: libaom0 (>= 1.0.0) but it is not going to be installed
                Depends: libc6 (>= 2.29) but it is not going to be installed
                Depends: libcairo2 (>= 1.2.4) but it is not going to be installed
                Depends: libcodec2-0.8.1 but it is not going to be installed
                Depends: libglib2.0-0 (>= 2.12.0) but it is not going to be installed
                Depends: libgsm1 (>= 1.0.18) but it is not going to be installed
                Depends: liblzma5 (>= 5.1.1alpha+20120614) but it is not going to be installed
                Depends: libmp3lame0 (>= 3.100) but it is not going to be installed
                Depends: libopenh264-5 (>= 2.0.0) but it is not going to be installed
                Depends: libopenjp2-7 (>= 2.0.0) but it is not going to be installed
                Depends: libopus0 (>= 1.1) but it is not going to be installed
                Depends: librsvg2-2 (>= 2.14.4) but it is not going to be installed
                Depends: libshine3 (>= 3.1.0) but it is not going to be installed
                Depends: libsnappy1v5 but it is not going to be installed
                Depends: libspeex1 (>= 1.2~beta3-1) but it is not going to be installed
                Depends: libtheora0 (>= 1.0) but it is not going to be installed
                Depends: libtwolame0 (>= 0.3.10) but it is not going to be installed
                Depends: libva2 (>= 2.2.0) but it is not going to be installed
                Depends: libvorbis0a (>= 1.1.2) but it is not going to be installed
                Depends: libvorbisenc2 (>= 1.1.2) but it is not going to be installed
                Depends: libvpx5 (>= 1.6.0) but it is not going to be installed
                Depends: libwavpack1 (>= 4.40.0) but it is not going to be installed
                Depends: libwebp6 (>= 0.5.1) but it is not going to be installed
                Depends: libwebpmux3 (>= 0.6.1-2) but it is not going to be installed
                Depends: libx264-155 but it is not going to be installed
                Depends: libx265-165 (>= 2.9) but it is not going to be installed
                Depends: libxvidcore4 (>= 1.2.2) but it is not going to be installed
                Depends: libzvbi0 (>= 0.2.35) but it is not going to be installed
                Depends: zlib1g (>= 1:1.2.0) but it is not going to be installed
 libavdevice58 : Depends: libasound2 (>= 1.0.16) but it is not going to be installed
                 Depends: libavc1394-0 (>= 0.5.3) but it is not going to be installed
                 Depends: libc6 (>= 2.28) but it is not going to be installed
                 Depends: libcaca0 (>= 0.99.beta17-1) but it is not going to be installed
                 Depends: libcdio-cdda2 (>= 10.2+0.94+2) but it is not going to be installed
                 Depends: libcdio-paranoia2 (>= 10.2+0.94+2) but it is not going to be installed
                 Depends: libdc1394-22 but it is not going to be installed
                 Depends: libdrm2 (>= 2.4.47) but it is not going to be installed
                 Depends: libgl1 but it is not going to be installed
                 Depends: libiec61883-0 (>= 1.2.0) but it is not going to be installed
                 Depends: libjack-jackd2-0 (>= 1.9.10+20150825) but it is not going to be installed or
                          libjack-0.125
                 Depends: libopenal1 (>= 1.14) but it is not going to be installed
                 Depends: libpulse0 (>= 0.99.1) but it is not going to be installed
                 Depends: libraw1394-11 but it is not going to be installed
                 Depends: libsdl2-2.0-0 (>= 2.0.9) but it is not going to be installed
                 Depends: libsndio7.0 (>= 1.1.0) but it is not going to be installed
                 Depends: libv4l-0 (>= 0.5.0) but it is not going to be installed
                 Depends: libx11-6 but it is not going to be installed
                 Depends: libxcb-shape0 but it is not going to be installed
                 Depends: libxcb-shm0 but it is not going to be installed
                 Depends: libxcb-xfixes0 but it is not going to be installed
                 Depends: libxcb1 but it is not going to be installed
                 Depends: libxext6 but it is not going to be installed
                 Depends: libxv1 but it is not going to be installed
 libavfilter7 : Depends: libass9 (>= 1:0.13.6) but it is not going to be installed
                Depends: libbs2b0 but it is not going to be installed
                Depends: libc6 (>= 2.29) but it is not going to be installed
                Depends: libflite1 (>= 1.4-release-9~) but it is not going to be installed
                Depends: libfontconfig1 (>= 2.12.6) but it is not going to be installed
                Depends: libfreetype6 (>= 2.2.1) but it is not going to be installed
                Depends: libfribidi0 (>= 0.19.2) but it is not going to be installed
                Depends: liblilv-0-0 (>= 0.14.2~dfsg0) but it is not going to be installed
                Depends: libmysofa0 (>= 0.7~) but it is not going to be installed
                Depends: librubberband2 but it is not going to be installed
                Depends: libva2 (>= 1.7.3) but it is not going to be installed
                Depends: libvidstab1.1 but it is not going to be installed
                Depends: libzmq5 (>= 3.2.3+dfsg) but it is not going to be installed
 libavformat58 : Depends: libbluray2 (>= 1:0.2.2) but it is not going to be installed
                 Depends: libbz2-1.0 but it is not going to be installed
                 Depends: libc6 (>= 2.28) but it is not going to be installed
                 Depends: libchromaprint1 (>= 1.3.2) but it is not going to be installed
                 Depends: libgme0 (>= 0.5.5) but it is not going to be installed
                 Depends: libgnutls30 (>= 3.6.6) but it is not going to be installed
                 Depends: libopenmpt0 (>= 0.3.0) but it is not going to be installed
                 Depends: libssh-gcrypt-4 (>= 0.8.0) but it is not going to be installed
                 Depends: libxml2 (>= 2.7.4) but it is not going to be installed
                 Depends: zlib1g (>= 1:1.2.0.2) but it is not going to be installed
 libavresample4 : Depends: libc6 (>= 2.17) but it is not going to be installed
 libavutil56 : Depends: libc6 (>= 2.29) but it is not going to be installed
               Depends: libdrm2 (>= 2.3.1) but it is not going to be installed
               Depends: libva-drm2 (>= 1.3) but it is not going to be installed
               Depends: libva-x11-2 (>= 1.3) but it is not going to be installed
               Depends: libva2 (>= 2.1.0) but it is not going to be installed
               Depends: libvdpau1 (>= 0.2) but it is not going to be installed
               Depends: libx11-6 but it is not going to be installed
 libpostproc55 : Depends: libc6 (>= 2.17) but it is not going to be installed
 libswresample3 : Depends: libc6 (>= 2.29) but it is not going to be installed
                  Depends: libsoxr0 (>= 0.1.0) but it is not going to be installed
 libswscale5 : Depends: libc6 (>= 2.29) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

JuserZhang, 2019-12-20:
默认的清华源找不到相应的安装包，添加debian的源就行了。

closed 2019-12-20 by JuserZhang

wanganjin233, 2019-12-21:
The following NEW packages will be installed:                                                                                       
  gcc-8-base:armhf libc6:armhf libgcc1:armhf libidn2-0:armhf libunistring2:armhf                                                    
WARNING: The following essential packages will be removed.                                                                          
This should NOT be done unless you know exactly what you are doing!                                                                 
  apt adduser (due to apt) gpgv (due to apt) libapt-pkg5.0 (due to apt) libc6 (due to apt) libgcc1 (due to apt)                     
  libgnutls30 (due to apt) libseccomp2 (due to apt) libstdc++6 (due to apt) base-files base-passwd                                  
  libdebconfclient0 (due to base-passwd) bash libtinfo6 (due to bash) debianutils (due to bash) bsdutils                            
  libsystemd0 (due to bsdutils) coreutils libacl1 (due to coreutils) libattr1 (due to coreutils) libselinux1 (due to coreutils)     
  dash dpkg (due to dash) debconf (due to dash) diffutils libbz2-1.0 (due to dpkg) liblzma5 (due to dpkg) zlib1g (due to dpkg)      
  tar (due to dpkg) e2fsprogs libblkid1 (due to e2fsprogs) libcom-err2 (due to e2fsprogs) libext2fs2 (due to e2fsprogs)             
  libss2 (due to e2fsprogs) libuuid1 (due to e2fsprogs) fdisk libfdisk1 (due to fdisk) libmount1 (due to fdisk)                     
  libncursesw6 (due to fdisk) libsmartcols1 (due to fdisk) findutils grep libpcre3 (due to grep) gzip hostname init                 
  systemd-sysv (due to init) init-system-helpers (due to init) perl-base (due to init-system-helpers) libc-bin login                
  libaudit1 (due to login) libpam0g (due to login) libpam-runtime (due to login) libpam-modules (due to login) mount                
  util-linux (due to mount) ncurses-bin sed sysvinit-utils libcap-ng0 (due to util-linux) libudev1 (due to util-linux)              
0 upgraded, 5 newly installed, 821 to remove and 0 not upgraded.                                                                    
Need to get 3,059 kB of archives.                                                                                                   
After this operation, 1,853 MB disk space will be freed.                                                                            
You are about to do something potentially harmful.                                                                                  
To continue type in the phrase 'Yes, do as I say!'     

这是为什么

dony71, 2020-03-06:
> 默认的清华源找不到相应的安装包，添加debian的源就行了。

how to do this?
I replace /etc/apt/source.list with debian source below, but still problem

deb http://deb.debian.org/debian buster main contrib non-free
deb-src http://deb.debian.org/debian buster main contrib non-free
deb http://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb-src http://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb http://deb.debian.org/debian buster-updates main contrib non-free
deb-src http://deb.debian.org/debian buster-updates main contrib non-free
deb http://deb.debian.org/debian buster-backports main contrib non-free
deb-src http://deb.debian.org/debian buster-backports main contrib non-free