openfans-community-offical/Debian-Pi-Aarch64 #75: Your connection is not private Attackers might be trying to steal your information

cxjava, 2019-12-23:
- Chrome version: `Version 79.0.3945.88 (Official Build) (64-bit)`
- OS version: `macOS  Catalina 10.15.2 (19C57)`
- Pi version: `2019-11-30-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-U1-Release-plus++.img`

When I access the web GUI `192.168.123.37:9090`, the following error is here, and can't find the `Proceed to 192.168.123.37 (unsafe)`

```
Your connection is not private
Attackers might be trying to steal your information from 192.168.123.37 (for example, passwords, messages, or credit cards). Learn more
NET::ERR_CERT_REVOKED

Help improve Chrome security by sending URLs of some pages you visit, limited system information, and some page content to Google. Privacy policy
192.168.123.37 normally uses encryption to protect your information. When Google Chrome tried to connect to 192.168.123.37 this time, the website sent back unusual and incorrect credentials. This may happen when an attacker is trying to pretend to be 192.168.123.37, or a Wi-Fi sign-in screen has interrupted the connection. Your information is still secure because Google Chrome stopped the connection before any data was exchanged.

You cannot visit 192.168.123.37 right now because its certificate has been revoked. Network errors and attacks are usually temporary, so this page will probably work later.

```

[image]

links:

[https://support.apple.com/en-us/HT210176](https://support.apple.com/en-us/HT210176)

[https://forum.openwrt.org/t/net-err-cert-invalid-from-google-chrome-in-macos-catalina/46435](https://forum.openwrt.org/t/net-err-cert-invalid-from-google-chrome-in-macos-catalina/46435)

Any suggestions?

aiminickwong, 2019-12-24:
Clear you brower's cache and cockie

closed 2019-12-24 by aiminickwong

cxjava, 2019-12-25:
no, I open it in an incognito window, still the same error. I guess the problem is `the TLS server certificates and issuing CAs using RSA keys must use key sizes greater than or equal to 2048 bits` 

check this link: https://support.apple.com/en-us/HT210176

so can we generate more strong RSA key in next release for webgui?