openfans-community-offical/Debian-Pi-Aarch64 #83: Impossibile to overclock Deepin-30-12-19 RPI4

MarcoPower01, 2020-01-03:
On the new deepin 30-12-19 is impossible to overclock when I overclock my rpi4 it crashes, on the old version of deepin 30-11-19 it didn't crash

P. S:I have 2 micro sd the first deepin30-12-19 and the second is deepin30-11-19

aiminickwong, 2020-01-03:
Please show your conf.

MarcoPower01, 2020-01-03:
my overclock config is this:
temp_limit=85
temp_soft_limit=85
arm_freq=2000
gpu_freq=650
sdram_freq=720
sdram_schmoo=0x02000020
dtparam=sd_overclock=100
over_voltage=6
over_voltage_sdram=8
sdram_over_voltage=3
over_voltage_sdram_c=6
over_voltage_sdram_i=4
over_voltage_sdram_p=4

MarcoPower01, 2020-01-03:
The problem is in kernel8.img if I swap the kernel8 file with the kernel8 (30-11) the System boot up but as I said the usb are not working.

aiminickwong, 2020-01-04:
you can try load uas kernel mod and try again.

MarcoPower01, 2020-01-04:
Yes but when I boot my system with the (30-11) kernel, 
I see this on boot: https://imgur.com/Qdyo8Vi

aiminickwong, 2020-01-04:
Do not swap boot

MarcoPower01, 2020-01-04:
Yes but If I don't swap the boot I can't overclock
As i said the overclock crash the last kernel8

MarcoPower01, 2020-01-04:
So if I want to use deepin 30-12-19 i can't overclock because it crash the system

aiminickwong, 2020-01-05:
cuz the kernel mod file version in system must as the same as file in boot partition.

MarcoPower01, 2020-01-05:
is the same i even reinstalled 2 times the os,there is a problem in the kernel you can try yourself if you want use the last deepin and use the same overclock config after tell me if it doesn't boot.

aiminickwong, 2020-01-06:
That's not overclock issue, it's normally simple kernel version's issue.

MarcoPower01, 2020-01-06:
I know is a kernel issue that happened only when I overclock but there is a fix out there?

aiminickwong, 2020-01-07:
the kernel's mod and the kernel must be as the same version

closed 2020-01-07 by aiminickwong

MarcoPower01, 2020-01-07:
So deepin 30-12-19.img have a kernel version different from what it should be. Yes I reinstalled again and again the os, so I was reflashing the entire micro sd a lot of time. But if I want to overclock (config.txt) it crash the sistem. OUT OF THE BOX I DIDN'T DO ANY MODIFICATION ON THE KERNEL. when this is going to be stable i just may retry to reinstall this os.