openfans-community-offical/Debian-Pi-Aarch64 #89: libc6-dev : Depends: libc6 (= 2.28-10) but 2.29-6 is to be installed

344389541, 2020-01-12:
pi@raspbian:~$ sudo apt-get install -f libc6-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc6-dev : Depends: libc6 (= 2.28-10) but 2.29-6 is to be installed
             Depends: libc-dev-bin (= 2.28-10) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
pi@raspbian:~$

344389541, 2020-01-12:
2019-12-30-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-U2-Release.img

aiminickwong, 2020-01-14:
Please read readme carefully.
https://github.com/openfans-community-offical/Debian-Pi-Aarch64#3-162-version-mismatch-when-installing-packages-using-apt-command

closed 2020-01-14 by aiminickwong