openfans-community-offical/Debian-Pi-Aarch64 #102: Where can I install the header package for the corresponding kernel?

hughjfchen, 2020-02-19:
The installed kernel version is 4.9.88, however, I can't find the corresponding headers package for this kernel version, hence I can't install DKMS kernel module. Where is the header package?

dony71, 2020-03-06:
same question
install "sudo apt-get install raspberrypi-kernel-headers" but not availble

hughjfchen, 2020-03-07:
Well, I can't figure out how to fix that yet, hence I switch back to raspbian with the 64-bit kernel.

dony71, 2020-03-10:
> Well, I can't figure out how to fix that yet, hence I switch back to raspbian with the 64-bit kernel.

where do you get raspbian arm64 image?

hughjfchen, 2020-03-11:
> > Well, I can't figure out how to fix that yet, hence I switch back to raspbian with the 64-bit kernel.
> 
> where do you get raspbian arm64 image?

Actually, it a raspbian 32-bit user-land with a 64-bit kernel and a 64-bit deiban within a nspaw container. Sokaki- has a repo which you can download such an image from: https://github.com/sakaki-/raspbian-nspawn-64

yi-ge, 2020-03-29:
I have the same question...

PHCSJC, 2020-04-29:
Same problem.
How install kernel headers?

aiminickwong, 2020-06-09:
Down load rpi officical  source code.

closed 2020-06-09 by aiminickwong

IS2511, 2020-08-09:
Doesn't really explain much...There are no source files in /lib/modules/kernel_version_here/build, how do I build a kernel module?