openfans-community-offical/Debian-Pi-Aarch64 #106: how to start by usb boot

williamnie, 2020-03-07:
pi 4b how to start by usb boot , I wrote the system in SD card and USB device separately，Mounted the USB device to the Raspberry Pi, and deleted partition 2 of the SD card, the system failed to start。
However, the official operation of the Raspberry Pi can be implemented in this way. 
How does this system work?

kppomzh, 2020-04-04:
I have the same problem again, and I also want to boot from USB completely independently. I wonder if there is such a solution?
It seems that an EEPROM mentioned that this is achievable, but now the method has not officially become a release feature.

williamnie, 2020-04-04:
> I have the same problem again, and I also want to boot from USB completely independently. I wonder if there is such a solution?
> It seems that an EEPROM mentioned that this is achievable, but now the method has not officially become a release feature.

you can see this link,but donot suppot this system, [https://www.quarkbook.com/?p=638](https://www.quarkbook.com/?p=638)

ctrai88, 2020-04-11:
I have the same problem again, However, the official operation of the Raspberry Pi can be implemented in this way.How does this system work?

ctrai88, 2020-04-11:
you can see this link。https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/

kppomzh, 2020-04-12:
@ctrai88 thanks,but some kind of USB driver,if they thrust on pi4 ,Debian can't normally enter "init 5" mode.
You can see [https://github.com/openfans-community-offical/Debian-Pi-Aarch64/issues/114](url)

ctrai88, 2020-04-12:
this link is null ? https://github.com/openfans-community-offical/Debian-Pi-Aarch64/issues/114

ctrai88, 2020-04-12:
USB Boot Instructions
1、We are going to burn a second identical copy of Raspbian to the SSD. 
pi@raspberrypi:~ $ sudo dd if=/dev/mmcblk0 of=/dev/sda
61820928+0 records in
61820928+0 records out
31652315136 bytes (32 GB, 29 GiB) copied, 1603.67 s, 19.7 MB/s

2、 sudo fdisk -l

Disk /dev/mmcblk0: 29.5 GiB, 31652315136 bytes, 61820928 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 61820927 61288448 29.2G 83 Linux

Disk /dev/sda: 119.2 GiB, 128000000000 bytes, 250000000 sectors
Disk model: KS-C7N          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 61820927 61288448 29.2G 83 Linux

3、 sudo fdisk /dev/sda

4、sudo blkid

pi@raspberrypi:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="97709164-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="97709164-02"
/dev/mmcblk0: PTUUID="97709164" PTTYPE="dos"
/dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="d34db33f-01"
/dev/sda2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="d34db33f-02"

5、Update /boot/cmdline.txt

6、We are now ready to test booting from the SSD. Restart your Pi by issuing a

sudo reboot

Unfortunately at this step the system will not work properly.

ctrai88, 2020-04-12:
USB Boot Instructions
1、We are going to burn a second identical copy of Raspbian to the SSD. 
pi@itcommander:~$  sudo dd if=/dev/mmcblk0 of=/dev/sda
61820928+0 records in
61820928+0 records out
31652315136 bytes (32 GB, 29 GiB) copied, 1603.67 s, 19.7 MB/s

2、 sudo fdisk -l

Disk /dev/mmcblk0: 29.5 GiB, 31652315136 bytes, 61820928 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 61820927 61288448 29.2G 83 Linux

Disk /dev/sda: 119.2 GiB, 128000000000 bytes, 250000000 sectors
Disk model: KS-C7N          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 61820927 61288448 29.2G 83 Linux

3、 sudo fdisk /dev/sda

4、sudo blkid

pi@itcommander:~$  sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="97709164-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="97709164-02"
/dev/mmcblk0: PTUUID="97709164" PTTYPE="dos"
/dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="d34db33f-01"
/dev/sda2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="d34db33f-02"

5、Update /boot/cmdline.txt

6、We are now ready to test booting from the SSD. Restart your Pi by issuing a

sudo reboot

Unfortunately at this step the system will not work properly.How does this system work?

ctrai88, 2020-04-12:
USB Boot Instructions
1、We are going to burn a second identical copy of Raspbian to the SSD. 
pi@itcommander:~$  sudo dd if=/dev/mmcblk0 of=/dev/sda
61820928+0 records in
61820928+0 records out
31652315136 bytes (32 GB, 29 GiB) copied, 1603.67 s, 19.7 MB/s

2、  Use fdisk to verify 
sudo fdisk -l

Disk /dev/mmcblk0: 29.5 GiB, 31652315136 bytes, 61820928 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 61820927 61288448 29.2G 83 Linux

Disk /dev/sda: 119.2 GiB, 128000000000 bytes, 250000000 sectors
Disk model: KS-C7N          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
Disklabel type: dos
Disk identifier: 0x97709164

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       532480 61820927 61288448 29.2G 83 Linux

3、 Verify  using blkid

sudo blkid

4、Change PARTUUID 

sudo fdisk /dev/sda

5、Let’s verify our change using blkid

sudo blkid

pi@itcommander:~$  sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="97709164-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="97709164-02"
/dev/mmcblk0: PTUUID="97709164" PTTYPE="dos"
/dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="9969-E3D2" TYPE="vfat" PARTUUID="d34db33f-01"
/dev/sda2: LABEL="rootfs" UUID="8f2a74a4-809c-471e-b4ad-a91bfd51d7c3" TYPE="ext4" PARTUUID="d34db33f-02"

6、Update /boot/cmdline.txt

sudo nano /boot/cmdline.txt

7、We are now ready to test booting from the SSD. Restart your Pi by issuing a

sudo reboot

Unfortunately at this step the system will not work properly.How does this system work?

aiminickwong, 2020-06-09:
Web need new erom, will be update soon.

closed 2020-06-09 by aiminickwong