openfans-community-offical/Debian-Pi-Aarch64 #119: The C compiler "/usr/bin/cc"  is not able to compile a simple test program.

wanggao1990, 2020-05-12:
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/pi/Payload-SDK/testDataTrans-crosscomplie/build/CMakeFiles/CMakeTmp

    Run Build Command:"/usr/bin/make" "cmTC_5001a/fast"
    /usr/bin/make -f CMakeFiles/cmTC_5001a.dir/build.make CMakeFiles/cmTC_5001a.dir/build
    make[1]: Entering directory '/home/pi/Payload-SDK/testDataTrans-crosscomplie/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_5001a.dir/testCCompiler.c.o
    /usr/bin/cc    -o CMakeFiles/cmTC_5001a.dir/testCCompiler.c.o   -c /home/pi/Payload-SDK/testDataTrans-crosscomplie/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_5001a
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5001a.dir/link.txt --verbose=1
    /usr/bin/cc      -rdynamic CMakeFiles/cmTC_5001a.dir/testCCompiler.c.o  -o cmTC_5001a
    /usr/bin/ld: cannot find Scrt1.o: No such file or directory
    /usr/bin/ld: cannot find crti.o: No such file or directory
    /usr/bin/ld: cannot find -lc
    /usr/bin/ld: cannot find crtn.o: No such file or directory
    collect2: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_5001a.dir/build.make:87: cmTC_5001a] Error 1
    make[1]: Leaving directory '/home/pi/Payload-SDK/testDataTrans-crosscomplie/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_5001a/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

xiadeye, 2020-05-13:
me too,how to fix?

wanggao1990, 2020-05-14:
> me too,how to fix?

This Pi arach64 Image just contain a runtime gcc/g++ complier.   The same error occurs  when jusy run  `gcc main.c -o main`  to printf “hello world.”

Just cross complie source code on other platform  and excute the binary file on PI system.

Or if u have any other better solution,  tell me pls.

aiminickwong, 2020-06-09:
@wanggao1990 @xiadeye
U're not right, 1'st pls understand sid-used command from our readme.
The system had not include libc-dev pkg by default, run command :

```
sid-used apt install libc6-dev-arm64-cross

root@raspbian:~# dpkg  -L libc6-dev-arm64-cross| grep -E "crtn|crti|Scrt1"
/usr/aarch64-linux-gnu/lib/Scrt1.o
/usr/aarch64-linux-gnu/lib/crti.o
/usr/aarch64-linux-gnu/lib/crtn.o

root@raspbian:~# cat test.c
#include <stdio.h>
int main(void)
{
  printf("Hello world!\n");
  return 0;
}
root@raspbian:~# gcc test.c
root@raspbian:~# ./a.out
Hello world!

``

The everything will be OK.

So, read readme file 1'st pls!!