openfans-community-offical/Debian-Pi-Aarch64 #125: Can I use icloud cloud disk to help the project to split downloads?

zhaokelei, 2020-06-16:
I found that icloud cloud disk can be downloaded quickly, and can be downloaded in different ways, including for Chinese users.

aiminickwong, 2020-06-16:
Share for everyone, anyway, anywhere, enjoy! @zhaokelei

zhaokelei, 2020-06-16:
https://www.icloud.com/iclouddrive/0XCOHNaasFSYZRD9-rZhjCbNw#2020-06-15-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-U3-Release

https://www.icloud.com/iclouddrive/0ajjHJCKj5UX2kcK2YVyhqlSw#2020-06-15-OPENFANS-Debian-Buster-Aarch64-ext4-v2019-2.0-U3-Release-plus

https://www.icloud.com/iclouddrive/0OcmvFXI2MsUB6j5mFu3w-pEQ#2020-06-15-OPENFANS-Debian-Buster-Desktop-Aarch64-ext4-v2019-2.0-U3-Release

zhaokelei, 2020-06-16:
I put three versions of the image file diffluence address here, which can be downloaded at any time.

aiminickwong, 2020-06-16:
@zhaokelei You can change and upload the newest desktop edition 2020-06-16-U3 , thank you.

zhaokelei, 2020-06-17:
好的

aiminickwong, 2020-06-17:
@zhaokelei 20200617 be comming...  :)

zhaokelei, 2020-06-19:
> @zhaokelei 20200617 be comming... :)

Updated

aiminickwong, 2020-06-22:
@zhaokelei 
TKY for Sharing! :)

aiminickwong, 2020-06-24:
@zhaokelei 20200622 Had upgrade for all IMGs

closed 2021-01-14 by aiminickwong