openfans-community-offical/Debian-Pi-Aarch64 #138: build the k8s master node failed

hua1010, 2020-08-15:
I need to do a k8s cluster training, but I found that there are some problems with the master node construction. 
Maybe the kernel version is too high or the necessary modules are not insmod. 
Compared with the official raspberry, it is found that it is normal. 
The following is an failed log:
error execution phase preflight: [preflight] Some fatal errors occurred:
        [ERROR SystemVerification]: failed to parse kernel config: unable to load kernel module: "configs", output: "modprobe: FATAL: Module configs not found in directory /lib/modules/5.4.47-OPENFANS+20200622-v8\n", err: exit status 1

Hope to get your help,thx

NJ2046, 2020-10-30:
the same to you

hua1010, 2020-10-30:
you can use the official image, good luck

NJ2046, 2020-10-31:
thk

aiminickwong, 2021-01-14:
We're not support configs modules, the better way is change the you deploy scripts

closed 2021-01-14 by aiminickwong