openfans-community-offical/Debian-Pi-Aarch64 #143: Try to use miraclecast as the client in the desktop version to initiate streaming to the TV .尝试在桌面版中，使用miraclecast作为客户端，向电视发起串流

hyphantom, 2020-09-11:
Try to use miraclecast as the client in the desktop version to initiate streaming to the TV .
尝试在桌面版中，使用miraclecast作为客户端，向电视发起串流

Unfortunately, the test showed that one failed
遗憾的是 测试显示有一个未通过

Unable to start miracle-wifid &
miracle-wifid & 无法启动

reference 参考：
https://github.com/albfan/miraclecast
https://github.com/albfan/miraclecast/issues/4

hyphantom, 2020-09-11:
miracle-wifid &  log：
ERROR: supplicant: wpas (pid:32534) failed unexpectedly, relaunching after short grace period.. (supplicant_failed() in ../../../src/wifi/wifid-supplicant.c:2331)
ERROR: supplicant: wpas (pid:32547) failed unexpectedly, relaunching after short grace period.. (supplicant_failed() in ../../../src/wifi/wifid-supplicant.c:2331)
ERROR: supplicant: wpas (pid:32548) failed again.. entering grace period, waiting 30s before relaunching (supplicant_failed() in ../../../src/wifi/wifid-supplicant.c:2335)

aiminickwong, 2021-01-14:
What's your means?

hyphantom, 2021-01-14:
> What's your means?

Try to use miraclecast as the client in the desktop version to initiate streaming to the TV .
尝试在桌面版中，使用miraclecast作为客户端，向电视发起串流

aiminickwong, 2021-01-16:
Can't you tell us you steps, pls

closed 2021-01-26 by aiminickwong