openfans-community-offical/Debian-Pi-Aarch64 #163: Can't find link to download kernel upgrade packe

TorstenH82, 2021-01-15:
Can't find kernel package on network disks. Please provide direct link for download.
Thank you very much

TorstenH82, 2021-01-15:
I'm unable to download files from your provided file hoster because registration is not possible for a German user. Could you please provide files on Dropbox?

aiminickwong, 2021-01-16:
@TorstenH82 We had update in MEGA share.
[MEGA](https://mega.nz/folder/coVQAaZR#ifOeikkhJpGYw8B7vvlDOg)

TorstenH82, 2021-01-16:
> @TorstenH82 We had update in MEGA share.
> [MEGA](https://mega.nz/folder/coVQAaZR#ifOeikkhJpGYw8B7vvlDOg)

Thank you very much! But I can't find Kernel upgrade package. Could you please upload the package to update to latest kernel?

aiminickwong, 2021-01-16:
Have Done! @TorstenH82

TorstenH82, 2021-01-16:
> Have Done! @TorstenH82

Could you please provide the upgrade to 2020-01-06-2.0-U6 5.4.83-OPENFANS+20210111-v8 because this is your suggested kernel?

Again thank you very much.

aiminickwong, 2021-01-17:
Sys Upgrade U just used apt upgrade, But New Featrues upgrade to newest system are not support yes, the newest kernel version is :5.10.6-ForTest-OPENFANS+20210112-v8 or 5.4.83-OPENFANS+20210111-v8; the 2020-01-06-2.0-U6 5.4.83-OPENFANS + 20210111-v8 upkgs had updated.

TorstenH82, 2021-01-17:
> Sys Upgrade U just used apt upgrade, But New Featrues upgrade to newest system are not support yes, the newest kernel version is :5.10.6-ForTest-OPENFANS+20210112-v8 or 5.4.83-OPENFANS+20210111-v8; the 2020-01-06-2.0-U6 5.4.83-OPENFANS + 20210111-v8 upkgs had updated.

Thank you very much.
It would be great if you could add MEGA to you standard file locations and provide the images and updatea there in future, too.

closed 2021-01-17 by TorstenH82

aiminickwong, 2021-01-17:
@TorstenH82 added