openfans-community-offical/Debian-Pi-Aarch64 #171: ping： network is unreachable

hclasmn, 2021-02-12:
[image]
[image]

It happens every two days。

aiminickwong, 2021-02-12:
Your nic wlan0 hadn't any ip addr, the network is unreachable, check you router or network device DHCP pool time range or other settings.

hclasmn, 2021-02-12:
> Your nic wlan0 hadn't any ip addr, the network is unreachable, check you router or network device DHCP pool time range or other settings.

Used normally all the time。

Restart and return to normal when it happened

closed 2021-02-12 by aiminickwong