openfans-community-offical/Debian-Pi-Aarch64 #175: ufw not work

Vastzhang97, 2021-03-10:
Hi，my pi4b iso is 2021-01-03-OPENFANS-Debian-Buster-Aarch64-ext4-v2021-2.0-U6-Release-plus++.img.xz,I install ufw for it and ufw deny 81,but i can still access my application on port 81,what wrongs?

[image]

[image]

aiminickwong, 2021-03-10:
we used iptables, not ufw.

Vastzhang97, 2021-03-10:
> we used iptables, not ufw.

Ok,iptables work.There are plans to support UFW in the future？

Vastzhang97, 2021-03-10:
I find the problem.Ufw is not work with docker container by default,you slove it do as [follow](https://www.techrepublic.com/article/how-to-fix-the-docker-and-ufw-security-flaw/)

closed 2021-03-10 by Vastzhang97