openfans-community-offical/Debian-Pi-Aarch64 #176: 最新版本U6的

SonnTakuu, 2021-03-27:


SonnTakuu, 2021-03-27:
另外，像说明那样的Android软件如何运行呢？linux不支持onenote，第三方客户端不支持arm64。

hmsjy2017, 2021-04-03:
可以使用 QEMU 模拟 Windows 10 ARM64，然后运行 x86 软件。
至于 Android 软件，可以通过 Anbox 运行。

SonnTakuu, 2021-04-14:
HDMI和3.5输出均只能输出单声道怎么办，是哪里设置的不对吗？

SonnTakuu, 2021-04-14:
好像没有QEMU和Anbox的安装针对树莓派4的教程，可以指导一下吗。

hmsjy2017, 2021-04-14:
安装 Windows 10 有两种方式：
- 实机安装：[树莓派4b安装Win10 ARM](https://www.bilibili.com/read/cv7745562/)
性能损耗小，部署简单，但 Wi-Fi 无驱动，可以用以太网来代替
- 虚拟机安装
  + [树莓派完美运行win10 arm「满足你的日常需求](https://blog.csdn.net/qq_41676577/article/details/112974605)（树莓派爱好者基地官方的教程，安装较简单）
视频：https://www.bilibili.com/video/av884150344
  + [树莓派通过KVM安装Win10 Arm64系统](https://talk.quwj.com/topic/1818)（另一个视频，安装较为复杂，但可定制度高）

无论是哪种方式，都不能使用 3D 加速。

基地的系统暂时不能运行 Android 软件，官方说未来会支持。目前可以通过 Anbox 实现运行 Android 软件，但只能安装arm64-v8a的应用。
[Anbox在ARM64架构Linux上运行Android程序](https://talk.quwj.com/topic/2205)
安装过程较为繁琐。

SonnTakuu, 2021-04-14:
基地系统声卡输出单声道如何解决呢，使用本系统以后居然可以流畅播放MA10P视频不卡了，但可惜声卡不知道如何调整为双声道；要是这个解决了，这个系统真的让树莓派无敌了。

iPhoneから送信

SonnTakuu, 2021-04-14:
呃，我询问的是openfans树莓派爱好者基地的64位debian系统，无论是HDMI还是3.5在设置中都只有单声道可选。

iPhoneから送信

我只用树莓派官方系统，没有这个问题。

hmsjy2017, 2021-04-14:
我不是树莓派基地的官方人员，不是很了解。你可以在他们的 QQ 群里面提问。

SonnTakuu, 2021-04-14:
哦哦，谢谢了，
以前在群里问过，好像没有人给出方案。

iPhoneから送信

hmsjy2017, 2021-04-14:
如果要运行 Android 应用，用 [Lineage OS](https://konstakang.com/devices/rpi4/) 体验可能会更好一些。唯一缺点是无法硬解视频。

hmsjy2017, 2021-04-14:
你可以在[趣小组](https://talk.quwj.com/)提问，也许有人会给出回答。

SonnTakuu, 2021-04-14:
是啊，无法硬解就很淦。。
我有一个nas，有安装jellyfin服务，jellyfin-mpv客户端是开源项目，但只有基地系统因为自带mpv所以可以硬解播放MA10p。
超频以后网页也基本没问题，平时学习用的anki居然也有linux版本，所以如果声音问题能解决就可以代替我98%的台式机用途。
如果虚拟的win10arm可以跑跑galgame那就可以代替99%了。瞬间树莓派就不吃灰了。

iPhoneから送信

hmsjy2017, 2021-04-14:
Windows 10 ARM64 已经支持 x64 模拟了，效率竟然比模拟 x86 都高。

aiminickwong, 2021-06-20:
> HDMI和3.5输出均只能输出单声道怎么办，是哪里设置的不对吗？

See [here](https://github.com/openfans-community-offical/Debian-Pi-Aarch64/blob/master/README.md#3-167-change-audio-mono-output-to-stereo-for-desktop-version-system).

closed 2021-06-20 by aiminickwong