openfans-community-offical/Debian-Pi-Aarch64 #187: How to install chromium?

misaka19465, 2021-05-09:
I removed the package chromium-browser, and installed chromium, but the chromium doesn't work. When I wanted to install chromium browser again, the apt said
```
没有可用的软件包 chromium-browser，但是它被其它的软件包引用了。
这可能意味着这个缺失的软件包可能已被废弃，
或者只能在其他发布源中找到

E: 软件包 chromium-browser 没有可安装候选
```
Can someone offer me the debs of chromium?

aiminickwong, 2021-05-09:
command :

```
apt update; apt-cache search chromium
```

to  search the package.

misaka19465, 2021-05-09:
I want to install `chromium-browser`, which is installed by default. The package `chromium` doesn't work, here is the screenshot
[image]

misaka19465, 2021-05-09:
```
There is no chromium-browser package available, but it is referenced by other packages.
This could mean that the missing package may have been deprecated,
Or it can only be found in other publishing sources

E: The chromium-browser package has no candidate for installation
```
This is the translation of the output of `apt install chromium-browser`

misaka19465, 2021-05-09:
@aiminickwong What's wrong with it?

misaka19465, 2021-05-09:
Everything works after install Vulkan driver

closed 2021-05-09 by misaka19465