openfans-community-offical/Debian-Pi-Aarch64 #201: Have a better solution about making using 5Ghz wifi available in China

qqice233, 2021-07-08:
I had met the problem that raspberry pi 4 wifi channel list for China (CN) on 5GHz band is wrong, which had been mentioned in #79 before. But instead of changing the region manually, why don't we replace the outdated firmware (/lib/firmware/brcm/brcmfmac43455-sdio.clm_blob) with a new one? I tried to replace it, and after a reboot I can reconnect to my 5Ghz wifi again. 
However, every update for the system (e.g. 5.10.44-Release-OPENFANS+20210620) will restore the file, and I have to replace it again. So is it possible to include the newer firmware in every update, so everyone doesn't need to do anything else to make using 5Ghz wifi available in China? The newer firmware is attached below (which is from [here](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=291609
))
[brcmfmac43455-sdio.zip](https://github.com/openfans-community-offical/Debian-Pi-Aarch64/files/6781671/brcmfmac43455-sdio.zip)

aiminickwong, 2021-07-08:
THE NEWEST SYSIN HAD INCLUDE THE NEWEST FW, CAN YOU CHECK AGAIN?

qqice233, 2021-07-08:
OK. Thanks for your reply.

aiminickwong, 2021-07-08:
I RE OPEN THIS ISSUE, UTIL THISN ISSUE FIXTED.
AND I ALSO UPLOAD THIS FW ZIP PKG **brcmfmac43455-sdio.zip** TO OUR GIT REPO AT 

 [*FW/firmware-nonfree/brcmfmac43455-sdio.zip*](https://github.com/openfans-community-offical/Debian-Pi-Aarch64/tree/master/FW/firmware-nonfree)

aiminickwong, 2021-07-10:
@qqice233
We launched a discussion meeting internally and conducted related tests. I am very happy to tell you that the newest upgrade package ( **upkg-2021-07-10-v2021-2.0-U6-Release.zip** ) has updated and added to the firmware.
Thanks for you again about the suggestions.

closed 2021-07-10 by aiminickwong