openfans-community-offical/Debian-Pi-Aarch64 #204: After xorg is installed, no display after reboot

huangqingchao, 2021-08-14:
Device: Pi 4B
OS Version: 2021-06-11-OPENFANS-Debian-Buster-Aarch64-ext4-v2021-2.0-U6-Release.img

After I update the system
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo reboot
```

I installed `xserver-xorg`, `mate-desktop-environment-core`, `lightdm` in turn.
But no display after reboot（Black screen, only one underline is flashing）

huangqingchao, 2021-08-14:
I tried to remove `lightdm`
After restart, I can enter command line mode
When I execute `startx`, I get the error "no screens found (EE)"

huangqingchao, 2021-08-14:
Xorg.0.log

pi@pi:~$ cat /var/log/Xorg.0.log
[   533.426] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[   533.431] Build Operating System: Linux 4.19.0-16-arm64 aarch64 Debian
[   533.433] Current Operating System: Linux pi 5.10.42-Release-OPENFANS+20210611-v8 #1 SMP PREEMPT Fri Jun 11 17:19:45 CST 2021 aarch64
[   533.433] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1280 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:6D:86:7A vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=daf1c5e3-02 rootfstype=ext4 elevator=deadline fsck.repair=yes net.ifnames=0 cgroup_enable=1 cgroup_memory=1 cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 zswap.enabled=1 zswap.zpool=z3fold zswap.compressor=lz4 zswap.max_pool_percent=25 rootwait
[   533.440] Build Date: 19 April 2021  09:34:38AM
[   533.442] xorg-server 2:1.20.4-1+deb10u3 (https://www.debian.org/support) 
[   533.444] Current version of pixman: 0.36.0
[   533.447] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   533.447] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   533.454] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Aug 14 23:37:57 2021
[   533.457] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   533.458] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   533.459] (==) No Layout section.  Using the first Screen section.
[   533.459] (==) No screen section available. Using defaults.
[   533.459] (**) |-->Screen "Default Screen Section" (0)
[   533.459] (**) |   |-->Monitor "<default monitor>"
[   533.460] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   533.460] (==) Automatically adding devices
[   533.460] (==) Automatically enabling devices
[   533.460] (==) Automatically adding GPU devices
[   533.460] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   533.460] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   533.460] 	Entry deleted from font path.
[   533.460] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[   533.460] (==) ModulePath set to "/usr/lib/xorg/modules"
[   533.460] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   533.461] (II) Loader magic: 0xaaaaea648e08
[   533.461] (II) Module ABI versions:
[   533.461] 	X.Org ANSI C Emulation: 0.4
[   533.461] 	X.Org Video Driver: 24.0
[   533.461] 	X.Org XInput driver : 24.1
[   533.461] 	X.Org Server Extension : 10.0
[   533.462] (++) using VT number 1

[   533.467] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_31
[   533.469] (II) no primary bus or device found
[   533.469] (II) LoadModule: "glx"
[   533.470] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   533.473] (II) Module glx: vendor="X.Org Foundation"
[   533.473] 	compiled for 1.20.4, module version = 1.0.0
[   533.473] 	ABI class: X.Org Server Extension, version 10.0
[   533.474] (==) Matched modesetting as autoconfigured driver 0
[   533.474] (==) Matched fbdev as autoconfigured driver 1
[   533.474] (==) Assigned the driver to the xf86ConfigLayout
[   533.474] (II) LoadModule: "modesetting"
[   533.474] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[   533.474] (II) Module modesetting: vendor="X.Org Foundation"
[   533.474] 	compiled for 1.20.4, module version = 1.20.4
[   533.474] 	Module class: X.Org Video Driver
[   533.474] 	ABI class: X.Org Video Driver, version 24.0
[   533.475] (II) LoadModule: "fbdev"
[   533.475] (WW) Warning, couldn't open module fbdev
[   533.475] (EE) Failed to load module "fbdev" (module does not exist, 0)
[   533.475] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   533.475] (WW) Falling back to old probe method for modesetting
[   533.476] (EE) open /dev/dri/card0: No such file or directory
[   533.476] (WW) Falling back to old probe method for modesetting
[   533.476] (EE) open /dev/dri/card0: No such file or directory
[   533.476] (EE) No devices detected.
[   533.476] (EE) 
Fatal server error:
[   533.476] (EE) no screens found(EE) 
[   533.476] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   533.476] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   533.476] (EE) 
[   533.483] (EE) Server terminated with error (1). Closing log file.

aiminickwong, 2021-08-15:
You used "apt-get dist-upgrade" to upgrade to the unstable version;  we are used stable version, you break the sys libs and drivers, we hadn't any test on unstable versions.

closed 2021-08-15 by aiminickwong