openfans-community-offical/Debian-Pi-Aarch64 #207: kex_exchange_identification: read: Connection reset by peer

evenlee, 2021-09-05:
Hi There

I tried 2021-06-11-OPENFANS-Debian-Buster-Desktop-Aarch64-ext4-v2021-2.0-U6-Release.img.xz on my pi 4, and I do not have a monitor attached, when I try to ssh to the server, i got the error in the title, and I flashed my sd card twice, still got same error,

When I tried to https:// ip.of.pi.4:9090, got and cannot find link page, please help, thanks.

aiminickwong, 2021-09-06:
What's you ssh client?

aiminickwong, 2021-09-06:
It's seems maybe you ip or network is not normally.

evenlee, 2021-09-06:
Thanks aiminic, I used both terminal and iterm on my mac, they have save issue, I can see the assigned ip in my router, and ping command works. my router uses 10.0.0.1/24 as ip address.

aiminickwong, 2021-09-07:
Would you please provide the size of the SD card and the model of the PI?

evenlee, 2021-09-07:
I am using Raspberry Pi 4 Model B Rev 1.1(4GB RAM) with 32GB SD card.

evenlee, 2021-09-09:
still not got the root cause, but got a workaround. I used respberry PI OS, and got same error. then I attached a monitor, and enabled the ssh by ui, and it works.

closed 2021-09-09 by evenlee