openfans-community-offical/Debian-Pi-Aarch64 #208: After the first successful installation, I cannot log in using the default account and password

Kirito-F, 2021-09-06:
The installation process is normal, but when the process ends, the login interface I see cannot log in with the default account and password. This is true both on the desktop and SSH.
The account I tried was pi and the password was raspberry.
[image]
[image]

aiminickwong, 2021-09-06:
Any image's version?  And check you img's md5 1st.

Kirito-F, 2021-09-06:
I used 2021-06-11-OPENFANS-Debian-Buster-Desktop-Aarch64-ext4-v2021-2.0-U6-Release.img.
And I just successfully verified MD5. Then I tried to reinstall it.
I noticed that during the installation process, I can successfully enter the desktop. I tried to click various setting interfaces for a short time, which is normal. But soon the system automatically restarted. As shown in the second picture, the screen was still black after waiting for half an hour. So I tried to power off and restart. After restarting, I entered the login interface and tried the default account and password. The problem remains the same.
[image]
[image]

aiminickwong, 2021-09-07:
Would you please provide the size of the SD card and the model of the PI?

aiminickwong, 2021-09-07:
Do not "tried to click various setting interfaces for a short time" before system auto reboot

aiminickwong, 2021-09-07:
And  what settings U did?

Kirito-F, 2021-09-07:
> Would you please provide the size of the SD card and the model of the PI?

The size of the SD card is 64 GB. The model of the PI is 4B Rev 1.2 .

Kirito-F, 2021-09-07:
> And what settings U did?

I didn't change any settings that time. I just tried to see if the interface can open normally.

Kirito-F, 2021-09-07:
I reinstalled the system again. There is such a prompt after the first boot.
[image]
I didn't do anything. Soon after, the system automatically restarted. After such an error, the restart process was interrupted.
[image]

closed 2021-09-09 by Kirito-F

Kirito-F, 2021-09-09:
Finally, I tried to replace another memory card, and the problem was solved, but the previous memory card also passed a simple read-write test. I can guarantee that the previous memory card was purchased through formal channels. I don't know if this is a compatibility problem, but the problem has been solved, and I decided not to go deep into it.
Here are some information:
The previous memory card was jointly signed by banq and JD. It has 64g storage space and the speed standards of V30, A1 and U3. I later tried an ordinary Kingston memory card, which also has 64g storage space, but the speed standard is only C10.