openfans-community-offical/Debian-Pi-Aarch64 #214: Read-only filesystem

nicholas-zww, 2021-11-24:
### Discussed in https://github.com/openfans-community-offical/Debian-Pi-Aarch64/discussions/213

<div type='discussions-op-text'>

<sup>Originally posted by **nicholas-zww** November 22, 2021</sup>
how to make a read-only file system? </div>

aiminickwong, 2021-11-25:
mount or edit fstab with ro option. (ro, means read-only).

nicholas-zww, 2021-11-25:
if the partition is mounted as read-only. We cannot write/save anything. What I am asking is something like file-system overlay, meaning that I will still save and modify the files, but of course, all changes wont be saved after reboot.

aiminickwong, 2021-11-27:
An easy way is use a  script control ro/rw, and the specific directory for writeable choose a separate partition mount is better.

closed 2021-12-05 by aiminickwong